ros-controls/ros2_control #1: Import ros2_control

Karsten1987, 2017-09-15:
Top Level issue referencing all pending PRs

bmagyar, 2017-09-19:
Thanks Karsten! All merged and ready to demo ;) 

Even though it is only demo mode I have one comment that should be sorted: `visibility_control.h` is duplicated in all packages. This is something that should make it into a core ROS2 package and probably even included with a `#include <ros2/ros2.h>`

bmagyar, 2017-09-19:
@Karsten1987 Please add a short summary & steps to compile and set up to the readme .

closed 2018-09-10 by Karsten1987

chapulina, 2018-11-12:
As indicated on the README, the code in this repository is still a proof of concept. It may be worth it ticketing issues listing what's left for this to be recommended for use. Will there be a Crystal release?

bmagyar, 2018-11-15:
We don't have anyone currently working on `ros2_control` but we'd gladly welcome merge requests or even a new maintainer for that role. I personally won't be able to dedicate time to it until next year.

GavinStrunk, 2018-11-16:
@bmagyar @chapulina Hey I have been looking at this repo thinking I would be interested in contributing and maintaining it.  I would definitely like some help in terms of defining a roadmap, because it is a little unclear to me what people would like out of this compared to ros_controls beyond using some of the new functionality in ros2. Do you have a recommendation on how people have setup this type of meeting in the past?

chapulina, 2018-11-16:
Maybe a discourse post with some ideas and asking for feedback from the community could be a good starting point. It would also be nice to describe the current state of the code and what's missing - maybe @Karsten1987 has the most context?

bmagyar, 2018-11-16:
We'd like to move much of the existing codebase under ros2_control. Can we
get a hangouts together with the three of us?

GavinStrunk, 2018-11-16:
Yea sounds good to me. I'm pretty flexible in the next week so if there is a day and time that works for both of you I can make it work. I'm on Eastern US time zone.