ros-controls/ros2_control #7: build error

znstj, 2018-01-12:
controller_parameter_server/parameter_server.hpp:28:40: error: ‘rclcpp::node’ has not been declared
 class ParameterServer : public rclcpp::node::Node

parthc-rob, 2018-06-04:
I have the same error when trying to build these packages using ament in my ROS2 workspace. 
@Karsten1987 please detail steps for a successful build.

znstj, 2018-06-12:
I change something,and build succes.
but this package only a demo about ros_control.
I can't use it do something.

Karsten1987, 2018-06-12:
Sorry for being late here. I'll have a look it ASAP. We recently changed the namespaces in `rclcpp` and `rclcpp::node::Node` became `rclcpp::Node`. I'll revise these packages.

parthc-rob, 2018-06-12:
Thanks for the pointer, please check out [this PR](https://github.com/ros-controls/ros2_control/pull/8)
Build still fails but for a different reason. I'm not sure what's going wrong. Sorry, this is my first PR, let me know of any improvements I can make.

znstj, 2018-06-14:
You can test  build only one package,and find bug:
ament build --only-packages hardware_interface.
and next controller_interface,controller_manager,controller_parameter_server,and test_robot_hardware.
if you cann't build that,give me you E-mail,I can send for you the package I fixed the bug.

parthc-rob, 2018-06-14:
@znstj Thanks, but to clarify, I've fixed the issue myself for ros2_control in ros-controls/ros2_control#8
 The current ros2_control + ros2_controllers packages were probably made for [`rclcpp 0.0.3`](https://github.com/ros2/rclcpp/commit/ed26865b7118a4789717212b04853775e59fc423) based on the timing of the commits.

>I change something,and build succes.
>but this package only a demo about ros_control.
>I can't use it do something.
I'm now trying to debug ros-control/ros2_controllers since that is the package you can <i>use</i> to do something with ros2_control
> EDIT: cleaned ament workspace and rebuilt, now I have build errors caused by rclcpp::Time::now() in -
>ros2_controllers PR - WIP

Karsten1987, 2018-06-20:
@parthc-rob I opened a PR to your PR in order to make it compile with the latest ROS2 master.
You may want to accept them in order to update your top-level PR, so that the one PR make it compile.

parthc-rob, 2018-06-20:
@Karsten1987 Thanks a lot! Updated my PR's.

Karsten1987, 2018-06-22:
I just merged the two open PRs. Thanks for contributing.

Please note, that this current master only builds against ROS2 master, built from source. 
There is a new ROS2 release coming up soon, from which one I'll start tagging and branching the two repos here as well.

closed 2018-06-22 by Karsten1987

parthc-rob, 2018-06-22:
Thanks for taking the PRs to completion! Hope to make some more soon.

Yeah I tried compiling with Ubuntu 16.04/ros2-ardent but noticed the commits you made were geared towards the 18.04/ros2-bouncy release. Docker containers will do the job for now.