ros-controls/ros2_control #10: How to use ros2_control?

qkx515, 2018-09-08:
I would like to try real-time control in ros2. The only thing I find about ros2 control is this repository. However, I cannot find anything that explain how to use this. Could any one please give me a little hint?

In ros2 demo about pendulum_control, it seemes that they wrote their own code. Does this mean we still don't have a ready-to-use ros2-control? Thank you very much! :-)

Karsten1987, 2018-09-10:
The ros2 realtime demo is not strictly related to this repo. That demo shows the new features of ROS2, i.e. custom/rt memory allocator and the intraprocess communication.

The ros2_control repos were mainly developed for a demo. So you're right that there is no ready-to-use ros2-control with the same capabilities as in ROS1. However, I still believe this is the right starting point here and I appreciate if you'd try it out and/or contribute.

I'll put together a readme ASAP.

qkx515, 2018-09-10:
A readme would be awesome! Thank you very much.

One thing I don't understand is that the developers seem to have tried to migrate ros_control.

It is said on [Beta3 Overview](https://github.com/ros2/ros2/wiki/Beta3-Overview) that they have tried
* Experimental rewrite of ros_control

But it seems that they stopped that progress somehow.

---

You must be a expert in ros_control. I would really appreciate it if you could look at [this issue](https://github.com/ros-controls/ros_control/issues/354) for me. Thank you in advance.

:-)

Best regards,
Kaixian QU

closed 2018-09-12 by qkx515