ros-controls/ros2_control #12: ros_canopen integration

samiamlabs, 2019-04-13:
Hi,

I'm working on porting [ros_canopen](https://github.com/samiamlabs/ros_canopen/tree/crystal-devel) to ROS2 and on adding CANopen 402 support to [ODrive](https://odriverobotics.com/).

The overall goal is to make a cheap open-source motor control solution for custom robots with a large number of brushless motors on the same CAN bus. Hopefully at a high enough frequency for dynamics based "operational space control" and things like that.

The next step for me is to port ```canopen_motor_node```. It is a hardware interface for ```ros_control``` at the moment.

I recently released a Unity3d based robot simulator for ROS2 ([unity_ros2](https://github.com/DynoRobotics/unity_ros2)) and I plan to make something similar to the ros_control <=> gazebo integration in ROS1 for that as well.

I would appreciate some more info about this project before I start coding integrations. Do you plan to make any updates to this project relatively soon? What works and what doesn't work at the moment?

bmagyar, 2019-04-15:
The steps will be slightly different for ROS and ROS2 as the latter is very much still in a proof of concept phase. On the ROS side you should be fine and I can definitely recommend using the gazebo plugin for inspiration.

This repo contains a reimplemented proof of concept of ros_control for ROS2. It should cover all the basic needs for a simulation, don't see any blockers for you.

Karsten1987, 2019-05-07:
I am closing this issue as I hope @bmagyar was answering your questions. Feel free to re-open it and discuss further if you've still got uncleared questions.

closed 2019-05-07 by Karsten1987

samiamlabs, 2019-05-07:
I have made some progress on the ros_canopen integration. Not sure if it makes sense to have this as an issue here though. Probably better to open more specific issues related to relevant repositories.

Will probably be adding support for effort and velocity command interfaces in my fork soon to support additional control modes available on CANopen 402 motor drivers. Do you prefer I make these types of breaking changes to the API on my fork and submit merge-requests or is it better to discuss options before I start coding? Is there no development planned for ros2_control in the foreseeable future besides what I'm doing?

Karsten1987, 2019-05-07:
I am happy to review any pull request. I believe it makes sense that you open the PR whenever you feel that it's in a some-what reviewable state and we take the discussion from there.

samiamlabs, 2019-05-07:
Ok, will do :)