ros-controls/ros2_control #17: JointStateController for Gazebo

rjshim, 2019-07-09:
Hi,

I would like to ask how to use joint_state_controller to position-control a robot on the Gazebo simulator. It will be very nice if there is a demo example I can refer to (like [this](http://gazebosim.org/tutorials/?tut=ros_control) in ROS1). 

Thank you in advance,
Ryan

Karsten1987, 2019-08-08:
There is currently no gazebo_ros_control plugin. I think the gazebo_ros_pkgs currently don't support that. 
@chapulina, is that right? (sorry for randomly tagging you)

rjshim, 2019-08-08:
@Karsten1987 

I posted this issue last month when I didn't know there is no substitute for it yet. I will wait for it to be ported and released. 

Regards,
Ryan

chapulina, 2019-08-08:
We're waiting on an official `ros2_control` release before porting `gazebo_ros_control` to ROS 2. It's my understanding that the current implementation is a [proof of concept](https://discourse.ros.org/t/ros-control-in-ros2-wishlist/7803/28), so we're waiting for something more stable.

Here's the issue tracking this on `gazebo_ros_pkgs`' side: https://github.com/ros-simulation/gazebo_ros_pkgs/issues/512

bmagyar, 2019-08-08:
I wouldn't advise to wait for anything brand new on that front soon. We are in the process of porting the satellite packages over and there's plenty of work done and to be done still. The main packages will take more time. I recommend bodging together some gazebo support for @Karsten1987 .

chapulina, 2019-08-09:
From #18, it sounds like the current codebase is going to be officially released soon. In that case, we could start working on Gazebo integration after on top of the release. Once the full rewrite comes up, we can look into integrating that as well.

rjshim, 2020-08-27:
@Karsten1987 @chapulina  @bmagyar

I think I better grasped how to use gazebo plugins now than last year. I will create another issue thread when I have more questions.
Thank you for all the comments help!
Ryan

closed 2020-08-27 by rjshim