ros-controls/ros2_control #18: When will the Dashing version of ros2_control be released?

KevinHwangAi, 2019-07-25:
I'm tring to use Ros2 dashing version to develop my robot. When will the Dashing version of ros2_control be released?

reso1, 2019-08-05:
Same question here.

Karsten1987, 2019-08-05:
Thanks for the reminder. I’ll take it on my todo list to branch off and release it by the end of the week :)

reso1, 2019-08-06:
> Thanks for the reminder. I’ll take it on my todo list to branch off and release it by the end of the week :)

Thanks and appriciate your work and the ROS2 team!

Karsten1987, 2019-08-08:
I've opened #19 which is WIP for dashing. I have to work on the tests (new parameter API etc) and actually would love to re-iterate it a bit longer to address some of the applied workarounds at the time of writing this code (bouncy).
So I am a bit torn to release as-is for dashing without being able to break API later on.

Karsten1987, 2019-08-20:
I've just re-iterated over the dashing parameter design in #19. 
The essential here is that the global ParameterServer is not being used but parameters set directly at each node, a.k.a controller.
@reso1 If you feel like, you can check out the two `dashing_update` branches and give it a shot.If not, I'd release a first version of this on Friday for dashing.

reso1, 2019-08-25:
> I've just re-iterated over the dashing parameter design in #19.
> The essential here is that the global ParameterServer is not being used but parameters set directly at each node, a.k.a controller. [...]

Thanks, Karsten. I'll try both versions, it is very helpful for me!

Karsten1987, 2020-02-06:
It's been too long, but the `dashing` branches for both, `ros2_control` and `ros2_controllers` are now up to date and compatible with the ros2 dashing release.
I've triggered a release for both as well, which should be online within the next dashing sync.

Please don't expect too much stability on this. We just recently resumed more attention to this project again so there will definitely be breaking changes for future releases.

closed 2020-02-06 by Karsten1987