ros-controls/ros2_control #21: As a ros2_control developer, I want to have CI for my PRs so I can maintain the quality and stability of my contributions

piraka9011, 2019-12-16:
## Design Requirements

* Add the `action-ros2-ci` Github action to run on pull requests and create a:
  * [ ] Build and test workflow
  * [ ] Linting workflow

gavanderhoorn, 2019-12-16:
I don't have any particular preference, but seeing as `ros-controls/ros_control` is using `industrial_ci`, I thought I'd mention the ROS 2 and Github Actions support of the refactored version.

Actions specifically mentioned here: [ros-industrial/industrial_ci/readme: Quick Start - For GitHub Actions](https://github.com/ros-industrial/industrial_ci/tree/master#for-github-actions).

piraka9011, 2019-12-16:
@gavanderhoorn Interesting, I did not realize they had ROS2 + GH action support. 
I don't how well their ROS2 distros are supported but I don't mind using that instead for consistency.

gavanderhoorn, 2019-12-16:
> I don't how well their ROS2 distros are supported

whose?

piraka9011, 2019-12-16:
> whose?

Sorry, I meant the `ros-industrial/industrial_ci`.

gavanderhoorn, 2019-12-16:
Afaik, all distros. It's docker based. But I'll wait for @ipa-mdl to comment on this.

ipa-mdl, 2019-12-16:
> Afaik, all distros

Yep, all distros.  Even the officially unsupported ones (indigo and newer).

gavanderhoorn, 2019-12-16:
> > Afaik, all distros
> 
> Yep, all distros. Even the officially unsupported ones (indigo and newer).

Also all ROS 2 distros, correct?

ipa-mdl, 2019-12-16:
> Also all ROS 2 distros, correct?

Of course :sunglasses: 
All ROS distros (ROS1 and ROS2) since indigo

bmagyar, 2019-12-17:
> Also all ROS 2 distros, correct?
>
> Of course 😎
> All ROS distros (ROS1 and ROS2) since indigo

Since its industrial-CI is docker-based, I suppose we can expect all the
Linux coverage however what about Windows builds? Given the amount of macro
magic we may need to add to cmake, we should validate on a Windows build
too.

bmagyar, 2020-02-03:
@piraka9011 can we please also include a source-based build of `ros2_control` in the `ros2_controllers` repo? The reason `ros_control` and `ros_controllers` have an "ros_control from source" stage in the CI is to ensure that changes to the framework were brought to the controllers too.

bmagyar, 2020-02-05:
I think it would be very beneficial for the future to split the build and test steps on the CI as it is currently in `ros_control`. If we ever have flaky tests (which is hard to avoid unless we are willing to put a lot of time into tweaking) it makes it much quicker to re-run as the build step can be skipped.

Separate build and test steps also allow maintainers to have a better result at first glance, a failing build is more critical than a failing test.

I'm bringing this up as after the changes on #44 the gh-actions CI still fails. 

Another question: can linting be a separate stage in the CI?

piraka9011, 2020-02-05:
> Another question: can linting be a separate stage in the CI?

Linting is currently its own workflow.

Separating build and test is non-trivial ATM since there's no mechanism to group GH actions together. `action-ros-ci` currently runs build and test to emulate the ROS2 build farm.
The MacOS issues in #44 however can be resolved and shouldn't be a blocker for not running tests. We can just have GH actions run on Ubuntu 18.04 in the meantime until that's resolved.

bmagyar, 2021-02-15:
We've had this for a while, closing...

closed 2021-02-15 by bmagyar