ros-controls/ros2_control #24: As a robotics developer, I want to control my robot using position, velocity, or torque so I can use different controllers

piraka9011, 2019-12-16:
## Design requirements

The implementation for an interface that can read/set the position/velocity/effort/torque of a robot can be found in the `hardware_interface` package.

This issue will be used to track whether there are certain features missing such as proper error handling, logging, or potentially new features such as acceleration.

youtalk, 2020-05-01:
Is someone tackling on this issue?

bmagyar, 2020-05-01:
No and @piraka9011 expressed that AWS' contributions will be much less from now on as they had to reprioritize. Feel free to take it.

youtalk, 2020-05-01:
I got it. I start contribution more.

bmagyar, 2021-02-15:
The main purpose of this ticket was fulfilled. There's one idea we captured on a new ticket. Closing...

closed 2021-02-15 by bmagyar