ros-controls/ros2_control #26: As a robotics developer, I want to specify the limits of my robot's joints so I can ensure safe operation of my robot

piraka9011, 2019-12-16:
## Design requirements

Create a `joint_limits_interface` that ensures joint commands never go over certain thresholds.

* [ ] Limits can be defined from parameters
* [ ] Limits can be defined programatically

## References

* ['ros-controls/ros_control`'s `joint_limits_interface`](https://github.com/ros-controls/ros_control/blob/melodic-devel/joint_limits_interface)
* [Windows STL conflicts with `std::min/max`](https://stackoverflow.com/questions/4913922/possible-problems-with-nominmax-on-visual-c/4914108#4914108). 
```cmake
# Potential solution with CMakeLists.txt
if(WIN32)
  add_definitions(-DNOMINMAX)
endif()
```

drewskoots, 2020-08-24:
Hi, I was just wondering there is any update as to when the ```joints_limit_interface``` will be added to ros2_control. The roadmap mentioned something about May 2020 and I'm waiting on its implementation to begin migrating my code. Any info would be greatly appreciated!

bmagyar, 2020-08-31:
Incoming on #135

bmagyar, 2021-02-15:
We've decided to move this functionality into `hardware_interface` as this code is extremely simple.

closed 2022-06-28 by bmagyar