ros-controls/ros2_control #27: Create Eloquent branch

bmagyar, 2020-01-13:


bmagyar, 2020-01-13:
FYI @jordan-palacios @piraka9011 @Karsten1987

Karsten1987, 2020-01-13:
that eloquent branch is already existing: https://github.com/ros-controls/ros2_control/tree/eloquent

closed 2020-01-13 by Karsten1987

bmagyar, 2020-01-14:
Yup, I've opened this issue, pushed the branch and tagged all of yous to notify :) thanks for closing

Karsten1987, 2020-01-14:
Just out of curiosity. Why did you create it? It’s currently on par with crystal and neither this one nor dashing has any updates.

bmagyar, 2020-01-14:
We'll put the rosin work here that are all potentially breaking changes, didn't want to disturb those more stable releases.

Karsten1987, 2020-02-06:
So I've restored a working version for `dashing` as well as `master`.
I am going ahead and delete the stale `eloquent` branch as all new work should be targeted against `master`.

The moment we decide to cut a release for `eloquent` we'll branch off from master.