ros-controls/ros2_control #33: As a hardware manufacturer, I would like to use combined_robot_hw to ship my RobotHW implementation working with my hardware

bmagyar, 2020-01-17:
# Design

Port over [combined_robot_hw](https://github.com/ros-controls/ros_control/tree/melodic-devel/combined_robot_hw)

# References

http://wiki.ros.org/combined_robot_hw

suab321321, 2020-01-25:
@bmagyar are these beginner issues?I have mediocre level c++ skills.

suab321321, 2020-01-25:
@bmagyar is there more detail how one should proceed into this?

piraka9011, 2020-01-25:
@suab321321 `combined_robot_hw` extends the [`hardware_interface`](https://github.com/ros-controls/ros2_control/tree/crystal/hardware_interface).
This involves moving all the ROS1 features and functionality to use the ROS2 API instead.

Per an earlier discussion with Bence tho, it seems that the hardware interface may change soon so it might better to help port over some of the controllers (like differential drive) instead or see #26 and #28.

suab321321, 2020-01-26:
@piraka9011 I have to implment the same thing but using ROS2 api isnt it?
First I will understand what `combined_robot_hw` is happening by looking at `ROS1` code,then I will implement the exact same thing but using `ROS2` api for making `publisher` and `subscriber`.Am I right?

piraka9011, 2020-01-26:
> First I will see what `combined_robot_hw` is doing by looking at its `ROS1` code,then I will implement the exact same thing but using `ROS2` api.

That would be great, and documenting what needs to be done in this issue would be must appreciated.

> but I need to wait for sometime because 2 new features are yet to merged made for `hardware_interface`.Am I right?

There are 2 PRs to fix the eloquent and dashing builds so yes, waiting for those to land (or branching from them) would be a good place to start from.

suab321, 2020-01-27:
Thanks @piraka9011 I will begin the work

suab321321, 2020-01-27:
Thanks @piraka9011.Soon I will begin the work.

suab321321, 2020-01-27:
@piraka9011 thanks.I will begin soon

destogl, 2020-03-05:
Hi all,
@bmagyar and @piraka9011 This would be a good point to discuss how `combind_robot_hw` should look like. Does this mean that combined robot has 2 robots or can also have some additional sensors/actors? This is an issue I have in  `ros_control`, therefore we introduced internally [CombinedRobotSensorHW](https://github.com/KITrobotics/ros_control/tree/combined_robot_sensor_hw/combined_robot_sensor_hw) Class.

Is it possible to consider this? For me is very hard to believe that I am the only one putting additional sensors and actors to a robot.

Thank You for Your effor and feedback :)

destogl, 2021-02-15:
We have this natively in the new architecture.

closed 2021-02-15 by destogl