ros-controls/ros2_control #45: Add clang-format file to repo

bmagyar, 2020-02-04:


JafarAbdi, 2020-02-21:
@bmagyar Is there any style that you would prefer (for example [MoveIt clang-format](https://github.com/ros-planning/moveit/blob/master/.clang-format) .? I could create a PR to fix this issue

bmagyar, 2020-02-21:
I think we'd be quite happy with most of that. One thing though that I remember seeing in moveit a lot is 
```
if (a)
  b();
```

while I'd prefer personally

```
if (a)
{
  b();
}
```
for more consistency. Can this be added?

JafarAbdi, 2020-02-21:
Seems like clang-format doesn't support this, but we could use clang-tidy to do so [see](https://clang.llvm.org/extra/clang-tidy/checks/readability-braces-around-statements.html)

bmagyar, 2020-02-26:
Could you please open a PR with this?

Karsten1987, 2020-03-24:
for an initial start for the clang format, does this help? https://github.com/ament/ament_lint/pull/226

This at least sorts the `#includes` correctly.

bmagyar, 2021-02-15:
opted for uncrustify to follow the crowd

closed 2021-02-15 by bmagyar