ros-controls/ros2_control #48: ros2_control and real-time requirements

carlossvg, 2020-02-06:
Hi `ros_control` folks, 

There were discussions in the real-time working group (RTWG) about ROS 2 projects working on real-time related applications. The name of this project was mentioned as some of the applications this project targets require real-time capabilities. 

It would be great to to have some feedback from this group, for example by providing use cases with hard real-time requirements, sharing your experience in the development of real-time applications and tools (i.e: [realtime_tools](https://github.com/ros-controls/realtime_tools)) or getting involved in the development or the testing of new features.

Some of the recurrent topics discussed in the RTWG are the rework of the rclcpp executor, the creation of a new C executor in [rclc](https://github.com/micro-ROS/rclc/tree/master/rclc) with deterministic behavior,  [tracing](https://micro-ros.github.io/docs/tutorials/advanced/tracing/), [performance testing](https://discourse.ros.org/uploads/short-url/k7liZMEZDFBXfvo2Fa2HTwLsAPa.pdf), etc. Feel free to join the RTWG and discuss any of these topics or propose new ones. 

You can read the minutes of previous meetings in discourse with the [wg-real-time](https://discourse.ros.org/tag/wg-real-time) tag and check upcoming events [here](https://index.ros.org/doc/ros2/Governance/).

Karsten1987, 2020-03-16:
@carlossvg thanks for reaching out. We (at least I) don't have any concrete feedback regarding realtime requirements other than that we have them ;-) 

I am very intrigued into leveraging the static executor for the controller manager. Ideally, we would like to use the C++ version of it, but the rclc might be more advanced than the static executor in rclcpp. 

I think the two WGs (real-time and ros-control) are very closely related in terms of their requirements when it comes to deterministic behavior. I would thus love to actually roll the ball back to you guys and ask for feedback on the [design docs for ros_control](https://github.com/ros-controls/roadmap). One precise PR which I believe could be of interest is this one: https://github.com/ros-controls/roadmap/pull/8/files

destogl, 2021-02-15:
We need to have tests on the runner provided by @olivier-stasse

closed 2021-02-15 by destogl