ros-controls/ros2_control #60: add ROS services for controller manager

Karsten1987, 2020-02-19:
There is currently no way to load controller dynamically at runtime through ROS2 services.

v-lopez, 2020-08-28:
I'll be looking into this in the next few weeks.

If possible i'll add the distinction between loaded and started controller, and between unloaded and stopped.

v-lopez, 2020-09-02:
I've begun working on this, implementing the loaded/started distinction.

Currently a controller is only `configured` when the controller manager is `configured` [here](https://github.com/ros-controls/ros2_control/blob/master/controller_manager/src/controller_manager.cpp#L110)
And is only `activated`  when the controller manager is `activated` [here](https://github.com/ros-controls/ros2_control/blob/master/controller_manager/src/controller_manager.cpp#L120)

Additionally I was gonna set it up that the controller is:
loadController -> `configures` when it is loaded, and `activated` on start.
Stop `deactivates`, unload `cleanups`
A controller is  `configured` when loading, or when the controller_manager is  `configured`

bmagyar, 2020-09-02:
Very good place to start a discussion on naming of controller states!

* loading in controller manager terms comprises of loading the actual controller plugin, initializing and configuring it.
* starting is the same as activating, I propose to rename this in ros2_control to activate
* stopping is the same as deactivating, let's rename to deactivate
* unloading is disposing of the loaded plugin instance after all cleanups were made. If we are to keep load, this should be kept too.

v-lopez, 2020-09-02:
It's interesting because unload doesn't do any unloading. It was one of our issues with ROS1 control, because we'd saved hours of testing if we could unload a library, build it with some changes, and load it again.

I don't know if unloading is possible and what side effects it could have. If it was possible, you'd only be able to unload if any other instances of plugins from that library were also unloaded beforehand. 
I don't think it's something we should focus on.

But the naming "unload" might be confusing.
I don't know if we should have:

- loadController
- configureController
- activateController
- deactivateController
- cleanupController

v-lopez, 2020-09-02:
Work in progress here: https://github.com/ros-controls/ros2_control/pull/139

So far going with old nomenclature, but that can be changed.

destogl, 2020-09-07:
> But the naming "unload" might be confusing, since it doesn't unload the library.

I think that having `unload` as opposite to `load` of a controller is OK. IMO `cleanup` would be even more confusing.

v-lopez, 2020-09-15:
https://github.com/ros-controls/ros2_control/pull/139 is ready for review, I'm still using old nomenclature.

destogl, 2020-12-12:
This is resolved in the current implementation. We are using LifecycleNode-state names and services exists and work.

closed 2020-12-12 by destogl