ros-controls/ros2_control #65: Joint Trajectory performance on servos

guru-florida, 2020-02-24:
How does the Joint Trajectory controller in this project perform for receiving position/vel/acc or position/effort messages and projecting them onto servo position waypoints? How does it control the effort input without knowing torque/current feedback?

I'm studying the code but based on the README this repo may not  be ready for normal use on ROS2? README says, 

> "...presented at ROSCON 2017 and is currently not under active development."

 I can't find any other existing Joint Trajectory Controllers besides MoveIt!2 perhaps.

Possibly related issues #10 
Possibly related ROS Answer - [What is the purpose of ros_control and ros_controller?](https://answers.ros.org/question/317060/what-is-the-purpose-of-ros_control-and-ros_controller/)

Thanks!

bmagyar, 2020-02-24:
Thanks for pointing out the part about the readme, we are in fact have been actively developing since last December. Here's a bit more on what to expect this year: https://github.com/ros-controls/roadmap

This time around documentation is part of the roadmap but unfortunately it's a little further down the line. In the mean time I believe @Karsten1987 had some older tutorial pages, or am I imagining things?

The "possibly related ROS Answer" post is horrible, it's the typical "here is my homework, please help" question where he didn't even realize that `ros_control` hosts the framework, `ros_controllers` hosts implemented controllers...

Indeed MoveIt2 is also expecting `ros2_control` to be fully available, the project is receiving some contributions from moveit developers as well.

As for the joint trajectory controller I am not sure what the question is but here's a simple example: a joint position is requested, the joint is set up with limits. A spline interpolation (there are a few available in the controller) is used to generate a one-dof joint trajectory. Joint positions are defined in angles for rotational and meters for linear actuators. This is transformed into a specific servo position by *your* RobotHW implementation, aka the Hardware Abstraction Layer. 
Here's an old (note, 2014!!) presentation which is still a good summary of the general idea and some internals: https://vimeo.com/107507546
Please for further questions try using ROS discourse, the issue tracking here is for development and bugs, not really for support.

guru-florida, 2020-02-25:
Thanks @bmagyar. I'm glad to hear this is being actively developed again. My question was just how far along this project is or if it was superseded. For example, I dont see the spline calculations. I've reviewed all the code in ros2_control and controllers and it seems pretty straight forward. I definitely don't want to be a "here is my homework" guy. ;) I reviewed the joint_trajectory controller and Trajectory classes and I dont see where it does spline interpolation. I see the Trajectory::sample(timestamp) method  but it returns an iterator which just gets the closest point from the Trajectory given from the msg (as passed via the lambda callback). I only see the 1 state and 1 trajectory controller in the repo.

Glad to hear ros2_control will be integrated with MoveIt!2. :) I've been looking at their latest release last week. I'm also working on using Crocoddyl optimal control.

closed 2020-02-25 by guru-florida

gavanderhoorn, 2020-02-25:
> The "possibly related ROS Answer" post is horrible, it's the typical "here is my homework, please help" question where he didn't even realize that `ros_control` hosts the framework, `ros_controllers` hosts implemented controllers...

So that's what I get for spending some time answering those questions .. :cry: 

> For example, I dont see the spline calculations. I've reviewed all the code in ros2_control and controllers and it seems pretty straight forward. I definitely don't want to be a "here is my [...]

That's perhaps just the case in the version that was ported to `ros2_controllers`. See [here](https://github.com/ros-controls/ros_controllers/blob/2d5ba0744a1eec7c9720ef3df6e8b571466d8f90/joint_trajectory_controller/include/trajectory_interface/quintic_spline_segment.h) for the `quintic_spline_segment.h` which certainly does interpolation (either linear, cubic or quintic).

This is used [here](https://github.com/ros-controls/ros_controllers/blob/2d5ba0744a1eec7c9720ef3df6e8b571466d8f90/joint_trajectory_controller/include/joint_trajectory_controller/joint_trajectory_controller_impl.h#L378) by the JTC.

bmagyar, 2020-02-25:
> So that's what I get for spending some time answering those questions .. cry

You did great ;) This unfortunately can't be said about the OP.

bmagyar, 2020-02-25:
Just to tie it back here, I've created a new document in the roadmap repo to collect good resources which we can use down the line for The Promised Documentation Pages.
https://github.com/ros-controls/roadmap/pull/5

guru-florida, 2020-02-25:
> > So that's what I get for spending some time answering those questions .. cry
> 
> You did great ;) This unfortunately can't be said about the OP.

Agreed. It was only because of your answer I linked it as possibly related. It wasn't because of the OP question.

guru-florida, 2020-02-27:
> That's perhaps just the case in the version that was ported to `ros2_controllers`. See [...]
 
Ok good, so I am not crazy. lol. I was aware ROS1 ros_control was much farther ahead. Roadmap helped too. I watched the 2014 you (@bmagyar) posted to, thanks! I'm trying to decide where to go from here. I would like to help port the more advanced trajectory controllers from the ROS1 code but I am also under a deadline to show this humanoid walking or at least balancing, and I'm up to my neck in optimal MPC whitepapers so I'm spread thin. Basically trying to do what was in the vimeo video at the end with the humanoids...that looks like optimal MPC but perhaps it's MoveIt OMPL. My joint state controller, IMU state and gravity compensation are already done, and my joints are compliant for human interaction/feedback, so I just need to start answering the JointTrajectory messages from my MPC controller and send to hardware. I wish I reviewed ros2_control in detail 2 months ago. I saw it but dismissed it because of the "not under active development" in the README and I was anticipating the coming of MoveIt!2 alpha. 

I am thinking I will do a quick and dirty TrajController now but I would like to come back and refactor my controllers as ros2_control controllers. I really like the new ros2_control and ControllerManager architecture.