ros-controls/ros2_control #66: Documentation for loading the controller is outdated

buschbapti, 2020-03-02:
Hello,

I am following the documentation to load a JointStateController. If I am not mistaken, the load_controller function as written in the README is outdated. The new version does not seem to take 3 string parameters but only 2 (name, type).

Moreover, it seems that the controllers are not under namespace ros_control but their own (joint_state_controller for JointStateController and joint_trajectory_controller for JointTrajectoryController). At least this is what I get by looking at the ros2_controller package (which address is also wrongly spelled in the README with controlllers instead of controllers).

The problem is that even when importing the joint_state_controller package I get loader for controller not found. For reference, I am using the following command:

`cm.load_controller("joint_state_controller",` "joint_state_controller::JointStateController");`

which might be wrong. Should the controller be loaded in a different way?

Thanks in advance.

Karsten1987, 2020-03-03:
That change got introduced in #41 
@jordan-palacios Do you mind coming up with a follow-up PR to address the documentation?

ddengster, 2020-03-04:
"joint_state_controller/JointStateController" should work

Karsten1987, 2020-03-16:
@jordan-palacios friendly ping.

buschbapti, 2020-03-25:
@ddengster Yes it does indeed. Thanks for the clarification.

rjshim, 2020-07-09:
@buschbapti 

Did it really work for you? I don't know what I did wrong. I get the
"Loader for controller 'joint_state_controller/JointStateController' not found" message.

rjshim, 2020-07-10:
For future users who might struggle because of the same reason as I had, try below, it worked. 
```sh
  cm.load_controller(
    "test_joint_state_controller",
    "test_controller/JointStateController");
```
You can also see sample examples at this [link](https://github.com/ros-controls/ros2_controllers/blob/master/joint_state_controller/test/test_load_joint_state_controller.cpp).

This was really confusing because the input **_"controller_type"_** for the function **_"load_controller"_** is the **_name of the plugin_** that you use, not the **_plugin type_**. I feel like there should be consistency in naming variables.

destogl, 2020-12-12:
There is an explanation in the rod-controls/ros2_control_demos repository README about this.

closed 2020-12-12 by destogl