ros-controls/ros2_control #70: Using quotes around ament package names in ament_target_dependencies()

bmagyar, 2020-03-19:
I found some inconsistencies in our cmake files regarding the usage of quotes around ament package names. I don't know where this comes from but I'd prefer to drop them and use quotes to highlight the semantics when the intention is defining an actual string.
If there is a strong *practical* argument for keeping them, let's make sure we have them but then IMHO we should write *every single occurence* of an ament package in quotes, not only in `ament_target_dependencies`.

Example from `ros2_control`, multiple styles in a single command:
```
add_library(
  hardware_interface
  SHARED
  src/joint_command_handle.cpp
  src/joint_state_handle.cpp
  src/operation_mode_handle.cpp
  src/robot_hardware.cpp
)
target_include_directories(
  hardware_interface
  PUBLIC
  include)
ament_target_dependencies(
  hardware_interface
  "rclcpp"
  "rcpputils"
)
```

Example from `ros2_controllers`:
```
  ament_add_gtest(test_trajectory_controller
    test/test_trajectory_controller.cpp
    ENV config_file=${CMAKE_CURRENT_SOURCE_DIR}/test/config/test_joint_trajectory_controller.yaml)
  target_include_directories(test_trajectory_controller PRIVATE include)
  target_link_libraries(test_trajectory_controller
    joint_trajectory_controller
  )
  ament_target_dependencies(test_trajectory_controller
    "controller_parameter_server"
    "control_msgs"
    "hardware_interface"
    "rclcpp"
    "rclcpp_lifecycle"
    "rcutils"
    "realtime_tools"
    "test_robot_hardware"
    "trajectory_msgs"
  )

  ament_add_gtest(
    test_load_joint_trajectory_controller
    test/test_load_joint_trajectory_controller.cpp
  )
  target_include_directories(test_load_joint_trajectory_controller PRIVATE include)
  ament_target_dependencies(test_load_joint_trajectory_controller
    controller_manager
    test_robot_hardware
  )

  ament_add_gtest(
    test_trajectory_actions
    test/test_trajectory_actions.cpp
  )
  target_include_directories(test_trajectory_actions PRIVATE include)
  target_link_libraries(test_trajectory_actions
    joint_trajectory_controller
  )
  ament_target_dependencies(test_trajectory_actions
    "controller_parameter_server"
    "control_msgs"
    "hardware_interface"
    "rclcpp"
    "rclcpp_lifecycle"
    "rcutils"
    "test_robot_hardware"
    "trajectory_msgs"
    "realtime_tools"
  )
```

bmagyar, 2020-03-19:
@jordan-palacios @piraka9011 @Karsten1987 yay or nay

piraka9011, 2020-03-19:
I prefer to follow convention of other ROS2 packages and have no quotes.

Karsten1987, 2020-03-19:
Certainly up for discussion, `rclcpp` does use quotes:
https://github.com/ros2/rclcpp/blob/master/rclcpp/CMakeLists.txt#L112-L122

I am fine with everything and would second @bmagyar thoughts to have a consistent approach within ROS2. I also like the thought that quotes are for actual strings (left aside that CMake anyway passes everything as strings), so this is a +1 for removing the quotes.

jordan-palacios, 2020-03-20:
+1 no quotes.

Karsten1987, 2020-03-20:
@bmagyar did I hear you volunteering for a PR to address this? That's awesome ;-)

bmagyar, 2020-03-23:
Yes indeed, please see the mentioned PR-s.

closed 2020-03-23 by bmagyar