ros-controls/ros2_control #72: Test dependency in metapackage

matthew-reynolds, 2020-03-25:
I'm not sure what a test dependency in a metapackage really does or offers, but I worry that it will pull in the dependency when using tools such as `rosinstall_generator` (Full disclaimer: I'm still very very green with ROS2, so I'm just pulling from ROS1 experience here).

Truthfully, I've never even seen a `test_depend` in a metapackage before. Can someone explain what the intended outcome of this dependency is? If there isn't one, I will open a PR to remove it in order to keep the dependencies of the metapackage slimmer (Similar to in ros_control, where tests are no longer part of the metapackage: https://github.com/ros-controls/ros_control/pull/411)

https://github.com/ros-controls/ros2_control/blob/9850a4bca817d34c776e35509e11474b3ea0d21f/ros2_control/package.xml#L20

bmagyar, 2020-04-06:
I don't know of any reason why we should keep it, in fact it feels like the same sort of setup that we had it ROS1. @Karsten1987 is there any particular reason for this, would something set on fire if we got rid of this test-dep?

@matthew-reynolds probably worth opening a PR and we can actually confirm things on the ROS2 buildfarm.

matthew-reynolds, 2020-04-09:
Closed in #75

closed 2020-04-09 by matthew-reynolds