ros-controls/ros2_control #73: Flaky ros2 master?

bmagyar, 2020-03-31:
@piraka9011 @Karsten1987 
Could you guys please help me out with where this originates from and how we could avoid it?

```
[ 98%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/transport/TCPChannelResourceSecure.cpp.o
  [ 99%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/transport/TCPAcceptorSecure.cpp.o
  [100%] Linking CXX shared library libfastrtps.so
  [100%] Built target fastrtps
  ---
  Aborted  <<< fastrtps
  --- stderr: fastrtps
  In file included from /usr/include/string.h:494:0,
                   from /home/runner/work/ros2_control/ros2_control/ros_ws/src/eProsima/Fast-RTPS/src/cpp/rtps/persistence/sqlite3.c:12709:
  In function ‘memcpy’,
      inlined from ‘vdbePmaReadBlob’ at /home/runner/work/ros2_control/ros2_control/ros_ws/src/eProsima/Fast-RTPS/src/cpp/rtps/persistence/sqlite3.c:88355:5,
      inlined from ‘vdbePmaReadVarint’ at /home/runner/work/ros2_control/ros2_control/ros_ws/src/eProsima/Fast-RTPS/src/cpp/rtps/persistence/sqlite3.c:88398:12:
  /usr/include/x86_64-linux-gnu/bits/string_fortified.h:34:10: warning: ‘__builtin_memcpy’: specified size between 18446744071562067968 and 18446744073709551615 exceeds maximum object size 9223372036854775807 [-Wstringop-overflow=]
     return __builtin___memcpy_chk (__dest, __src, __len, __bos0 (__dest));
            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  ---
  
  Summary: 83 packages finished [6min 41s]
    1 package failed: rcl_logging_log4cxx
    1 package aborted: fastrtps
    47 packages had stderr output: ament_cmake ament_cmake_copyright ament_cmake_core ament_cmake_cppcheck ament_cmake_cpplint ament_cmake_export_definitions ament_cmake_export_dependencies ament_cmake_export_include_directories ament_cmake_export_interfaces ament_cmake_export_libraries ament_cmake_export_link_flags ament_cmake_flake8 ament_cmake_gmock ament_cmake_gtest ament_cmake_include_directories ament_cmake_libraries ament_cmake_lint_cmake ament_cmake_pep257 ament_cmake_pytest ament_cmake_python ament_cmake_ros ament_cmake_target_dependencies ament_cmake_test ament_cmake_uncrustify ament_cmake_version ament_cmake_xmllint ament_lint_auto ament_lint_common fastrtps foonathan_memory_vendor gmock_vendor gtest_vendor launch_testing_ament_cmake python_cmake_module rcl_logging_log4cxx rmw_connext_cpp rmw_connext_shared_cpp rmw_implementation_cmake rosidl_adapter rosidl_cmake rosidl_generator_c rosidl_generator_dds_idl rosidl_parser rosidl_typesupport_connext_c rosidl_typesupport_connext_cpp rosidl_typesupport_interface test_interface_files
```

piraka9011, 2020-03-31:
We've been seeing this in our builds as well.
Yes, ros2 master isn't stable. Its a side effect of ros2 supporting multiple RMW implementations.

Karsten1987, 2020-03-31:
is that a GitHub action? I haven't seen this on the ROS2 buildfarm. What does that mean it's not stable? All the nightlies are compiling.

piraka9011, 2020-03-31:
I guess what I meant is the master branch for ros2 repos aren't always stable (builds have failed before).
This probably was caused by a change in a down stream dependency and needs a re-run.
The actions don't run at the same time/frequency as the buildfarm so that's why this might have not appeared there.

bmagyar, 2020-04-01:
This is a nightly too, should we schedule it to run right after the ROS2 master nightly to catch the stable state? :D 

What's strange to me is I remember seeing the exact failure in the past too...

bmagyar, 2020-04-01:
Can we schedule the same thing to run on travis at the same time? May shed some light on it the next time it happens.

bmagyar, 2020-04-09:
@Karsten1987 @piraka9011 should we report these failures?

from this morning: https://github.com/ros-controls/ros2_control/runs/566901267
```
 Traceback (most recent call last):
    File "<string>", line 1, in <module>
  ModuleNotFoundError: No module named 'numpy'
  CMake Error at cmake/rosidl_generator_py_generate_interfaces.cmake:212 (message):
    execute_process(/usr/bin/python3 -c 'import
    numpy;print(numpy.get_include())') returned error code 1
  Call Stack (most recent call first):
    /home/runner/work/ros2_control/ros2_control/ros_ws/install/ament_cmake_core/share/ament_cmake_core/cmake/core/ament_execute_extensions.cmake:48 (include)
    /home/runner/work/ros2_control/ros2_control/ros_ws/install/rosidl_cmake/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:286 (ament_execute_extensions)
    CMakeLists.txt:49 (rosidl_generate_interfaces)
```

bmagyar, 2020-04-21:
https://github.com/ros-controls/ros2_control/runs/604520000
```
   Makefile:140: recipe for target 'all' failed
  make: *** [all] Error 2
  ---
  --- stderr: rclcpp_lifecycle
  /home/runner/work/ros2_control/ros2_control/ros_ws/src/ros2/rclcpp/rclcpp_lifecycle/src/lifecycle_node.cpp: In constructor ‘rclcpp_lifecycle::LifecycleNode::LifecycleNode(const string&, const string&, const rclcpp::NodeOptions&)’:
  /home/runner/work/ros2_control/ros2_control/ros_ws/src/ros2/rclcpp/rclcpp_lifecycle/src/lifecycle_node.cpp:65:40: error: no matching function for call to ‘rclcpp::node_interfaces::NodeBase::NodeBase(const string&, const string&, rclcpp::Context::SharedPtr, const rcl_node_options_t&, bool)’
         options.use_intra_process_comms())),
                                          ^
  In file included from /home/runner/work/ros2_control/ros2_control/ros_ws/src/ros2/rclcpp/rclcpp_lifecycle/src/lifecycle_node.cpp:30:0:
  /home/runner/work/ros2_control/ros2_control/ros_ws/install/rclcpp/include/rclcpp/node_interfaces/node_base.hpp:40:3: note: candidate: rclcpp::node_interfaces::NodeBase::NodeBase(const string&, const string&, rclcpp::Context::SharedPtr, const rcl_node_options_t&, bool, bool)
     NodeBase(
     ^~~~~~~~
  /home/runner/work/ros2_control/ros2_control/ros_ws/install/rclcpp/include/rclcpp/node_interfaces/node_base.hpp:40:3: note:   candidate expects 6 arguments, 5 provided
  make[2]: *** [CMakeFiles/rclcpp_lifecycle.dir/src/lifecycle_node.cpp.o] Error 1
  make[1]: *** [CMakeFiles/rclcpp_lifecycle.dir/all] Error 2
  make[1]: *** Waiting for unfinished jobs....
  make: *** [all] Error 2
  ---
  Failed   <<< rclcpp_lifecycle	[ Exited with code 2 ]
  --- output: hardware_interface
```

destogl, 2020-12-12:
This should be resolved with #265 or am I wrong?

bmagyar, 2020-12-12:
Yep, also we aren't building against ros2 master anymore

closed 2020-12-12 by bmagyar