ros-controls/ros2_control #74: Foxy release

bmagyar, 2020-04-06:
List of required features from `ros2_control` and `ros2_controllers` for the first Foxy release:
* The `joint_trajectory_controller` from ROS Melodic with all it's features
* The `diff_drive_controller` from ROS Melodic with all it's features
* `gazebo_ros_control`

Waiting for contributions to this list @jordan-palacios  @Karsten1987  @piraka9011

bmagyar, 2020-04-06:
My first list is what I think of as bare minimum. 

I'd opt for shipping something now to have *some* functionality available but with ideally it'd be good to be able to ship some framework improvements (aka all the stuff in design docs) to Foxy too, meaning they may be slightly breaking changes, at least within the scope of framework components which we'd manage. The yearly release proposed starting from Foxy doesn't leave us with very good alternatives if we want to ship stuff in a timely manner.
What are your thoughts on this guys? @Karsten1987 @jordan-palacios @piraka9011

piraka9011, 2020-04-06:
SGTM, not sure how large of a change `gazebo_ros_control` would be but maybe focusing on framework/infra stuff might be better?

destogl, 2020-04-09:
@bmagyar, what is the best way to manage the roadmap for this? I mean the list of tasks with technical / implementation steps. We need to coordinate ourselves somehow...

Should we just open one issue, with a checklist of tasks and responsible person for it? Or ist it better to use GitHub-Project option? Has anyone experience with that? If we go the latter path, then I would need access rights to this repository.

bmagyar, 2020-04-09:
How about a new section in the roadmap repo for a general outline? I can look into how we could leverage github project for the execution

piraka9011, 2020-04-09:
@bmagyar @destogl We have a zenhub board setup in the roadmap repo. It does require an extension though. A GitHub project board also works.

ruffsl, 2020-04-16:
> A GitHub project board also works.

+1 for project visibility and being co-located with the repo. I was checking up on `gazebo_ros_control`, which led me here, wondering about the state of `ros2_control`, and where the project roadmap was.

destogl, 2020-04-19:
@ruffsl you can find it in [roadmap repository](https://github.com/ros-controls/roadmap)

destogl, 2020-11-18:
The PRs for the release:

[ros2_control](ros-controls/ros2_control):
* #234 
* #236 
* #240 

[ros2_controllers](ros-controls/ros2_controller):
* [joint_state_controller](ros-controls/ros2_controller#110)
* [forward_command_controller](ros-controls/ros2_controller#99)
* joint_trajectory_controller

[ros2_control_demos](ros-controls/ros2_controller_demos)
* [RRBotSystemHardware](ros-controls/ros2_controller_demos#37)

Demo Robots (at least one)
* ABB Studio
* ABB Robot
* KUKA Robot

gavanderhoorn, 2020-11-19:
> Demo Robots (at least one)
> 
>     * ABB Studio
>     * ABB Robot
>     * KUKA Robot

I haven't been able to attend the meetings, so this could have been discussed there.

Could you provide some more info on availability of interfaces for the two robot OEMs you mention?

destogl, 2020-11-20:
There are currently none, but people said they can provide them. I can notify you when we start to work on it.

I am not experienced with ABB.
For KUKA we will be porting RSI-Driver from kuka_experimental.

bmagyar, 2020-12-05:
I've added doc and source jobs to Foxy for ros2_control

https://github.com/ros/rosdistro/pull/27622

destogl, 2020-12-05:
How should we provide doc? Or how do we want to provide it? also in the long-term.

bmagyar, 2020-12-05:
I'd like to contribute those to the main ROS2 docs page if possible, we could create a sub-section there or we could even ask for a subdomain like https://navigation.ros.org/

destogl, 2020-12-05:
It would be great if we could have something directly in the repositories. So that each package has `doc` folder. I believe this would be the easiest to maintain and keep track of changes. What do you think?

bmagyar, 2020-12-07:
I was actually thinking about this last night and had a similar idea that we should try to keep them local *but* since it'd also be beneficial to have this appear in a central place, we could use a script that collates the individual package docs.

destogl, 2020-12-10:
How other people are doing this? I saw the recommendation that doc could be in the code repository. Is buildfsrm maybe doing this automatically?

bmagyar, 2021-01-01:
https://github.com/ros/rosdistro/pull/27872
https://github.com/ros/rosdistro/pull/27873

bmagyar, 2021-01-05:
https://discourse.ros.org/t/announcing-ros2-control-for-foxy/18274

closed 2021-01-05 by bmagyar