ros-controls/ros2_control #79: Some thoughts on controller and filters

destogl, 2020-04-24:
Hi,

in the calls, some of you mention about value limiters when talking about controller composition. What I understand, you would like to have a controller-like structure, which can limit/change values. If so, then this simple kine of controllers are actually filters. Would it be sensible to separate these two semantically, but keep the same parent class? The execution and structure would be actually the same. In this case, we would be able to dynamically compose filters and controllers which would be very nice. This is of course also possible without separation, but it could get confusing for the users. The same issue is with "controllers" which are actually only publishers in ros-control.

What do you think about that? @bmagyar @Karsten1987 @piraka9011 @olivier-stasse

olivier-stasse, 2020-04-25:
Could you elaborate on kine of controllers I was not able to get the code you are referring to.

For me ros2_control is going in the direction of avoiding to have too much semantic to make possible various low-level control and various high level control. IMHO semantic of the controllers should be put in ros_controllers or another repos, but *NOT* in ros2_control.

For instance the  ros_controllers which are used with Tiago are following a design pattern where all the controllers are semantically decoupled. We have be able in my group to have a complex graph-based, whole-body controller by having one ros1_controller using all the resources at once. Thanks to the agnostic status of ros1_control we can switch to both design patterns without any changes. I believe that this feature should be kept in ros2.

I am totally supporting the proposal of @Karsten1987 to have a more complex controller_manager component-based (but would suggest a graph structure), but I would not put this at the level of ros3_control.  They are too many software solutions, and committing to one specifically at the level of ros2_control seems to me not being the proper place. Probably you would like to have this discussion in the roadmap repo.

destogl, 2020-05-05:
I am not sure if it is possible to separate controller structure from `ros2_control` since controller manager should have knowledge about it. But then `controller_manager` can be placed in `ros2_controllers` package...

My proposal is to have different semantic regarding controller and filters. They have the same basic structure, but the meaning is different. `ros_control` is agnostic to semantic, but than, everything in it is a controller, e.g. components which only publish data. This is something I dislike. 

@olivier-stasse what do you man with "graph-based" or "graph structure"?

destogl, 2021-02-15:
This is obsolete so closing

closed 2021-02-15 by destogl