ros-controls/ros2_control #93: Check is cmake shouldn't have explicit find_package for gtest and gmock, prefer gmock

bmagyar, 2020-06-01:
Examples: 

* here we grab both gtest and gmock, one should really be enough the the gtest tests ported to gmock: https://github.com/ros-controls/ros2_control/blob/master/controller_manager/CMakeLists.txt#L47

* here no testing package was grabbet yet is magically works: https://github.com/ros-controls/ros2_control/blob/master/hardware_interface/CMakeLists.txt#L53 probably better to grab it explicitly

v-lopez, 2020-06-04:
**hardware_interfaces** uses [ament_lint_auto_find_test_dependencies](https://github.com/ament/ament_lint/blob/master/ament_lint_auto/cmake/ament_lint_auto_find_test_dependencies.cmake#L36)

Which gets the `ament_cmake_gtest` dependency from the package.xml
The `find_package` it performs doesn't have the REQUIRED flag.  So I believe we should add it explicitly.

Regarding **controller_manager**, `find_package(ament_cmake_gmock REQUIRED)` does not trigger a  `find_package(ament_cmake_gtest)`.

If you remove `find_package(ament_cmake_gtest)` from CMakeLists.txt and `<test_depend>ament_cmake_gtest</test_depend>` from package.xml, CMake execution will fail because `ament_add_gtest` command is not defined.

So IMO we need to keep `find_package` for `ament_cmake_gtest` and `ament_cmake_gmock`. 

Unless we want to move this discussion to a higher place where this behavior can be changed.

v-lopez, 2020-06-04:
Thinking a bit more about `ament_lint_auto_find_test_dependencies`, I thought it could've been done to avoid all this redundancy between package.xml and CMakeLists.txt.

But checking some core packages, they have the `ament_lint_auto_find_test_dependencies` as well as `find_package(ament_cmake_gtest REQUIRED)`.
For instance: [rclcpp](https://github.com/ros2/rclcpp/blob/master/rclcpp/CMakeLists.txt) and [rclcpp_action](https://github.com/ros2/rclcpp/blob/master/rclcpp_action/CMakeLists.txt)

bmagyar, 2020-06-04:
I also think we shoudl only use either gtest or gmock in the repo, but *at least* in a single package for simplicity and cleanliness.

bmagyar, 2021-02-15:
we are using gmock in all the packages AFAIK but it could be worth a check

bmagyar, 2023-05-29:
@GreatAlexander te gusta?

destogl, 2023-05-30:
@bmagyar is this actually still an issue?

christophfroehlich, 2025-03-13:
@AdityaPawar162 do you maybe want also to take this one?

soham2560, 2025-03-31:
@AdityaPawar162 let me know if you are working on this, or I can take this up

AdityaPawar162, 2025-04-01:
Hi @soham2560,
Feel free to take up this issue! Apologies @christophfroehlich, I missed your response earlier. I’ve resolved a similar issue in [control_toolbox](https://github.com/ros-controls/control_toolbox/pull/300)—you might find it helpful to check it out.

soham2560, 2025-04-01:
Works! @christophfroehlich let me know if I can take this up or should we reserve it for first timers

christophfroehlich, 2025-04-01:
as nobody picked it up for the past 5 years, you are welcome to fix it ;)

PR #2162 (2025-04-03): Make all packages use gmock, not gtest
## Brief
In this PR
- Moved all Test Packages to `gmock` instead of `gtest and/or gmock` (refer 7895b47273c161be225ddc8ef7274ecf88afa5ad)
    - Fixes #93 
    - used reference from https://github.com/ros-controls/control_toolbox/pull/300
- Fixed order of includes as suggested in above PR to match with [this](https://google.github.io/styleguide/cppguide.html#Names_and_Order_of_Includes) (refer 91c3aa9 and ab62583)

## How was this tested?
```bash
colcon build --symlink-install
source install/setup.bash
colcon test --packages-select controller_interface con [... cut]
closed 2025-04-03 by christophfroehlich