ros-controls/ros2_control #94: Implement `controller_manager::checkForConflict()` and implement resource management for controllers

bmagyar, 2020-06-01:


destogl, 2020-06-17:
This would be probably part of the `Actuators`, since it is only critical when controllers is doing to write simultaneously.

bmagyar, 2020-06-20:
I disagree. The way ros_control handles this in ROS1 is that one joint
command interface can only be used by one controller at a time. This is
decided at load time. You cannot have multiple controllers trying to write
to the same joint loaded and running at once. It is not a concurrency issue
but rather a resource-management one.

destogl, 2020-06-20:
@bmagyar I agree completely, that is what I meant, but my comment was not clear.

The point was, this is only relevant for actuators since generally sensors can be read from an arbitrary number of controlled. It would be very cool to achieve this and also important for cascade control and similar structures.

bmagyar, 2020-09-21:
@v-lopez it seems you've covered some of this already? at least in terms of scaffoldings

v-lopez, 2020-09-22:
The scaffoldings yes, the whole checking for conflict not though. I was under the impression that this was going to be handled with the new resource interface.

bmagyar, 2020-09-22:
I've added this to the ros1 features and also the new components projects... I think I'll remove it from the ros1 board as it's really only relevant for components at this point

destogl, 2021-02-05:
Although this is done for CM

#322 addressed this in ROS1 terms. There is currently only "notify" - "doSwitch" functionality. We should probably add also "trySwitch" functionality. Let's discuss it there further.

bmagyar, 2021-02-15:
Merging this with #315

closed 2021-02-15 by bmagyar