ros-controls/ros2_control #95: Move standard names for interfaces to control_msgs

bmagyar, 2020-06-02:
... once stable in hardware_interface/include/hardware_interface/control_type.hpp

destogl, 2020-06-02:
Should we gather proposals somewhere?

bmagyar, 2020-06-03:
On this ticket?

destogl, 2020-06-09:
OK, let's start then:

* Position
* Velocity
* Acceleration
* Effort
* Force

bmagyar, 2020-06-09:
What do you think about adding units to these? Of course taking linear/rotary joints into account so there could be 1-2 options

destogl, 2020-06-11:
Do you mean something like:
- Position-rad
- Position-m
- Velocity-rad/s
- velocity-m/s

bmagyar, 2020-06-13:
Ah sorry I wasn't clear on that. I'd add some documentation to these in the messages. Not to the names/types.

destogl, 2020-06-17:
I did something in this direction [here](https://github.com/ros-controls/ros2_control/blob/aa885a25f991b1f6e729cc46d4426342b3f31494/robot_control_components/include/robot_control_components/ros2_control_types.h#L35)

destogl, 2021-02-03:
I think we agreed not to do this anymore since using a simple header file with constants has a smaller footprint and dependencies than messages.

@bmagyar do you agree? if so let's close this.

bmagyar, 2021-02-03:
agreed!

closed 2021-02-03 by bmagyar