ros-controls/ros2_control #105: Is it sensible to use lifecycle noded to manage Robots, Sensors and Actuators

destogl, 2020-06-16:
Each component in ros2_control should have a strictly defined state, so that we enable more complex scenarios, like tool changers, stoping robots without killing nodes, etc.

This would also improve troubleshooting and IMHO more robust and transparent management of hardware.

I propose the following states:
```
using component_state_type = uint;
static constexpr component_state_type ROS2C_COMPONENT_STATE_LOADED = 1;
static constexpr component_state_type ROS2C_COMPONENT_STATE_CONFIGURED = 2;
static constexpr component_state_type ROS2C_COMPONENT_STATE_INITIALIZED = 3;
static constexpr component_state_type ROS2C_COMPONENT_STATE_STARTED = 4;
static constexpr component_state_type ROS2C_COMPONENT_STATE_STOPPED = 5;
static constexpr component_state_type ROS2C_COMPONENT_STATE_HALTED = 6;
```

@Karsten1987 Can we achieve this also with lifecycle components? Does it make sense to them here? Would this be something similar as you proposed for the controllers?

Karsten1987, 2020-06-16:
I've tried to mention this in [here](https://github.com/ros-controls/ros2_control/pull/80#issuecomment-642338013) but I strongly advice against making each actuator/sensor a node. There's quite a bunch of overhead coming with it (a.k.a. built-in services etc) which I personally don't see feasible. A second aspect is that these things are ideally as small as possible, not only to reduce their logical footprint, but also to guarantee a decent realtime behavior. Therefore, I'd rather see that each _component_ stays as much as possible ROS agnostic.

I agree though that it makes sense to provide some basic lifecycle functionality, at least some sort of activate/deactivate to be able to determine the lifecycle on runtime.

destogl, 2020-06-16:
Ok, than I will add states from above (maybe not all of them), so one can at least follow it from controller_manager.

closed 2020-06-16 by destogl