ros-controls/ros2_control #106: Make Robot class as main Container used from controller_manager

destogl, 2020-06-16:
After a discussion with @Karsten1987 yesterday, after the full integration of functionalities into Robot class as the main container for Sensors, Actuators, and hardware interfaces, we need to change interfaces of controller_manager and potentially controller to access use Robot and not RobotHardware class.

For more details see #101

olivier-stasse, 2020-06-16:
Thanks @destogl for starting this and taking the first hot comments.

A first lean implementation of Robot, Sensor and Actuators as in your diagram :
https://github.com/destogl/roadmap/blob/ros2_hardware_interface_design_proposal/design_drafts/images/ros2_control_core_diagram.svg
would be fantastic.

Let me know how it is possible to help.

gavanderhoorn, 2020-06-16:
Unfortunately I cannot attend the WG meetings, so I have to ask "stupid" questions here in issue comments: why was the name Robot chosen?

The use of `robot` in URDF and xacro has caused much confusion over the years, with users assuming they could / should not use the technology for anything not a robot (but still something which could be modelled using URDF).

Should this be avoided here?

Karsten1987, 2020-06-16:
We've come up with a diagram as follows. The idea would be to feature some sort of container instance, which has the capabilities to load - as in ROS1 - a stock robot, e.g. an industrial arm, a mobile base or humanoid. However, the container shall be generic enough to load individual hardware components such as actuators or sensors. We've talked about supporting things like conveyer belts through a set of individual actuators.  

[image]

As for the name, I am very open for inspiration: I am not settled on either, "robot" and "hardware [interface]".

destogl, 2020-09-11:
after ros-controls/ros2_control#121 not relevant any more

see also ros-controls/roadmap#22

closed 2020-09-11 by destogl