ros-controls/ros2_control #111: ros2_control usage example

rjshim, 2020-06-26:
Hi @Karsten1987 

I am working on porting source codes for manipulator robots written in ROS1 to ROS2. I am trying to implement things that written on this [instruction](http://wiki.ros.org/urdf/XML/Transmission), but I cannot find any resources for implementation in ROS2 that I can refer to. I can't even find anything related to transmission_interface/SimpleTransmission in this repository. Are there any instructions or sample examples that you are using when porting this ros_control repository?

Thank you for help in advance! 
Ryan

bmagyar, 2020-06-26:
Our readme was written in 2017. The project has been evolving recently quite a lot but we are far from feature parity with ROS1 ros_control. There are no reference implementations either, perhaps the closest you could find is [gazebo_ros2_control](https://github.com/ros-simulation/gazebo_ros2_control/pulls) or the simplistic setup in `test_robot_hardware`.

Transmissions are currently not in working order but being actively worked on, indeed SimpleTransmission is something I'm working on at the moment ;) 
You can check progress here: https://github.com/orgs/ros-controls/projects/4

rjshim, 2020-06-26:
> Transmissions are currently not in working order but being actively worked on, indeed SimpleTransmission is something I'm working on at the moment ;)

Glad to hear that. Thank you for the answer.

closed 2020-06-26 by rjshim