ros-controls/ros2_control #115: if you don't specify the parameter name in the cpp, the linter won't cry about the unused parameter and we save on some boilerplate

bmagyar, 2020-06-30:
if you don't specify the parameter name in the cpp, the linter won't cry about the unused parameter and we save on some boilerplate

```suggestion
  hardware_interface_ret_t configure(const ComponentInfo & /* actuator_info */)
  {
    return hardware_interface::HW_RET_OK;
  }
```

_Originally posted by @bmagyar in https://github.com/ros-controls/ros2_control/pull/101_

bmagyar, 2020-07-01:
@Karsten1987 @jordan-palacios @v-lopez @destogl please weigh in on this here.

My vote is with commenting out the argument name, I think it is a much cleaner way of denoting not wanting to use an argument. It is also one line less, overall reducing litter in the code.

v-lopez, 2020-07-01:
Comment out or remove altogether from the .cpp and leave on the hpp only.

Karsten1987, 2020-07-01:
I actually don't remember why we do this all over the place within the ros2 stack. I thought to remember that one of the linters (cppcheck, cpplint, uncrustify) would complain about it. If that's not the case any longer, I am okay with having the argument commented out.

jordan-palacios, 2020-07-01:
Agreed with the previous responses, commenting out the argument is the way.

destogl, 2020-12-12:
So it seams we are rather going for removing the arguments. see [comment](https://github.com/ros-controls/ros2_controllers/pull/125#discussion_r541755377)

bmagyar, 2021-02-15:
Done!!

closed 2021-02-15 by bmagyar