ros-controls/ros2_control #118: Tests on RT-PREEMPT

olivier-stasse, 2020-07-10:
### This issue is to track the implementation of a github-action runner using Linux RT-PREEMPT

The main issue with this thread is to implement a test to check if the main ros_control part are breaking minimal real-time requirements. 

Example: For torque control robot, a minimal 1 KHz loop is required. It also implies having at least 800 us free for computation over 1 ms. This time is typically what is required to solve a Quadratic Program with a robot of 30 DoFs with various local control with sensor feedback. The middleware foot print has to be minimal, i.e. not more than few tens of microseconds on a normal laptop.

### TO DO list
- [X] Formal support to have a serveur on the host institution (LAAS/CNRS)
- [X] Getting the computer
- [X] Focal install
- [ ] Linux 5.4 rt install
- [ ] Declare the github-action runner
- [ ] Writing the test using specifically the RT-PREEMPT call and testing the middleware

olivier-stasse, 2020-07-10:
Some real-time results on the computer with a Linux v5.4.47 kernel with the RT_PREEMPT patch:
```sh
sudo ./cyclictest -a -t -n -p99
[sudo] password for ostasse: 
# /dev/cpu_dma_latency set to 0us
policy: fifo: loadavg: 1.16 1.25 0.57 1/472 1496           

T: 0 ( 1491) P:99 I:1000 C: 103515 Min:      3 Act:   13 Avg:   12 Max:      30
T: 1 ( 1492) P:99 I:1500 C:  69010 Min:      3 Act:   12 Avg:   12 Max:      40
T: 2 ( 1493) P:99 I:2000 C:  51757 Min:      3 Act:   13 Avg:   12 Max:      45
T: 3 ( 1494) P:99 I:2500 C:  41406 Min:      3 Act:   12 Avg:   13 Max:      40
```

olivier-stasse, 2020-07-14:
Test on setting a self-hosted runner done.

olivier-stasse, 2020-07-20:
@bmagyar: To use the computer for the repository or the organisation (it would be probably better to add the runner only to the repository for a test) the instructions to go through are the following:
```
Setting -> Actions -> Add Runner.
```
The interface propose to choose the operating system ('Linux' is needed) and the architecture ('X64' is needed)

During the creation (and *ONLY* during) the interface provides instructions to install the services on the runner. I need you to send me the instructions.

Let me know when it is convenient for you to do this.

bmagyar, 2020-08-01:
@olivier-stasse I've sent an email with the instructions to you to your @laas.fr email, thanks for following up on this :+1: 

Also, you may find these useful/inspiring for some test implementations: https://github.com/ros2/realtime_support

olivier-stasse, 2020-08-05:
Dear @bmagyar on my side the runner is working and ready.
Let me know if it is not working for you.

destogl, 2021-02-15:
This is connected #48 and it replaces is actually.