ros-controls/ros2_control #132: Update component parsing to cover changes to min-max

bmagyar, 2020-08-21:
https://github.com/ros-controls/roadmap/issues/23

AndyZe, 2020-09-23:
Working on this

AndyZe, 2020-09-26:
@bmagyar @destogl Does component_info.cpp need to be updated as well, or is it correct now? Here's the data structure there:

```
/**
 * \brief This structure stores information about components defined for a specific hardware
 * in robot's URDF.
 */
struct ComponentInfo
{
  /**
   * \brief name of the component.
   */
  std::string name;
  /**
   * \brief type of the component: sensor or actuator.
   */
  std::string type;
  /**
   * \brief component's class, which will be dynamically loaded.
   */
  std::string class_type;
  /**
   * \brief name of the command interfaces that can be set, e.g. "position", "velocity", etc.
   * Used by joints.
   */
  std::vector<std::string> command_interfaces;
  /**
   * \brief name of the state interfaces that can be read, e.g. "position", "velocity", etc.
   * Used by Joints and Sensors.
   */
  std::vector<std::string> state_interfaces;
  /**
   * \brief (optional) key-value pairs of components parameters.
   */
  std::unordered_map<std::string, std::string> parameters;
};
```

destogl, 2020-10-21:
I reopen this since the issue is not settled completely (see [comment](https://github.com/ros-controls/ros2_control/pull/167#pullrequestreview-512395931))

Karsten1987, 2020-10-21:
I am btw strongly against CamelCase [here](https://github.com/ros-controls/ros2_control/blob/891f524ee4a9f9cc7c7ddccfb078db8b23435c41/hardware_interface/src/component_parser.cpp#L35-L36) and would vote for `snake_case`. I don't see any help in searching for "commandInterfaceType". First off, I believe the tag name is too long, but then the parser should at least allow for variants of this such as `commandinterfacetype`, `CommandInterfaceType`, `commandInterfacetype`, etc.

destogl, 2020-10-21:
> I am btw strongly against CamelCase [here](https://github.com/ros-controls/ros2_control/blob/891f524ee4a9f9cc7c7ddccfb078db8b23435c41/hardware_interface/src/component_parser.cpp#L35-L36) and would vote for `snake_case`.

I agree. We should update the example descriptions in the roadmap repo and then also the parser, as soon as we merge ros-controls/roadmap#27

PR #167 (2020-10-20): Add a struct for Interface information, update the test URDF
Closes #132
closed 2020-11-17 by bmagyar