ros-controls/ros2_control #138: Industrial CI from ROS main failing consistently

bmagyar, 2020-09-02:
On new PR we had to disable "Industrial CI from ROS main" due to consistent failure none of us were able to reproduce with (supposedly) identical installations.

 https://github.com/ros-controls/ros2_control/pull/136

The error was

```
  Starting function 'build_target_workspace'
  Starting >>> hardware_interface
  --- stderr: hardware_interface
  In file included from /opt/ros/foxy/include/control_msgs/msg/dynamic_joint_state.hpp:9,
                   from /root/target_ws/src/ros2_control/hardware_interface/include/hardware_interface/robot_hardware.hpp:22,
                   from /root/target_ws/src/ros2_control/hardware_interface/src/robot_hardware.cpp:15:
  /opt/ros/foxy/include/control_msgs/msg/detail/dynamic_joint_state__traits.hpp:27:25: error: expected initializer before '<' token
     27 | inline const char * name<control_msgs::msg::DynamicJointState>()
        |                         ^
  make[2]: *** [CMakeFiles/hardware_interface.dir/build.make:115: CMakeFiles/hardware_interface.dir/src/robot_hardware.cpp.o] Error 1
  make[1]: *** [CMakeFiles/Makefile2:166: CMakeFiles/hardware_interface.dir/all] Error 2
  make[1]: *** Waiting for unfinished jobs....
  make: *** [Makefile:141: all] Error 2
  ---
  Failed   <<< hardware_interface [9.54s, exited with code 2]
```

@Karsten1987 @piraka9011 do you guys know by any chance what this error may originate from?

gavanderhoorn, 2020-09-02:
@ipa-mdl: would you happen to have run into this already?

`dynamic_joint_state__traits.hpp` looks to be a template specialisation.

ipa-mdl, 2020-09-02:
This is not an issue of industrial_ci, but with the current `ros:foxy-ros-core` Docker image.
It is not possible to build `ros2_control` in it.

ipa-mdl, 2020-09-02:
It looks like the [last sync](https://discourse.ros.org/t/new-packages-for-foxy-fitzroy-2020-08-27/16126) is did not make it to Docker hub yet

bmagyar, 2020-09-21:
@ipa-mdl has there been an update with the osrf images? Should I just try again?

PR #158 (2020-09-22): run apt update && upgrade
fixes #138
closed 2020-09-22 by bmagyar