ros-controls/ros2_control #141: Add ResourceManager class to controller_manager_package

bmagyar, 2020-09-12:


destogl, 2020-09-12:
Tasks:

- [ ] Make maps with access to state and command interfaces of components check in the Ressource manager they are done properly, See example 6 where ActuatorHW provides commands and SensorHW provide state for the same Joint Component

destogl, 2020-11-18:
After we merge #236 this could be triggered again. There are no circular dependencies anymore.

Karsten1987, 2020-11-18:
I think we should close this. I was thinking about this a little bit in more detail and the resource manager should always be passed into the controller manager. The reason for this is that other applications such as the gazebo ros control packages are instantiating the resource manager "statically" with their internal URDF and don't rely on the controller manager to load plugins. 
Also semantically, I believe that the resource manager takes well place inside the hardware interface, as it's essentially only dealing with hardware components and has no strong dependencies to controllers and such.

destogl, 2020-11-19:
I agree!

@bmagyar: If Karsten's explanation is OK for you, please closeit.

closed 2020-12-12 by destogl