ros-controls/ros2_control #151: Unify return_type nomenclature in hardware_interface and controller_interface

v-lopez, 2020-09-17:
I propose the name change to keep the same nomenclature as for `hardware_interfaces`. What do you think?

```suggestion
  return controller_interface::return_type::OK;
```

_Originally posted by @destogl in https://github.com/ros-controls/ros2_control/pull/139#discussion_r489326777_

hardware_interface uses `OK` and controller_interface uses `SUCCESS`

bmagyar, 2020-09-17:
I agree and I think we should pick the shorter one, I vote for `OK`.

destogl, 2020-09-17:
I vote also foe OK. It is a bit broader term, but still with the clear meaning.

Karsten1987, 2020-09-17:
OK sounds good to me, too.

v-lopez, 2020-09-18:
All right, I'll do after https://github.com/ros-controls/ros2_control/pull/139 is merged to avoid conflicts

destogl, 2021-02-26:
So we go with `OK`. Rename `return_type::SUCCESS` everywhere to `return_type:OK`.  But keep `SUCCESS` where it is defined and make its value equal to `OK`. Mark `SUCCESS` is deprecated.

PR #374 (2021-04-07): Replace controller_interface::return_type SUCCESS by OK and mark SUCCESS as deprecated
As per the title.

Fix #151
closed 2021-04-07 by bmagyar