ros-controls/ros2_control #155: We broke MoveIt builds

destogl, 2020-09-20:
With the merge of #134 the MoveIt source build fails. Check:  ros-planning/moveit2#283

Since we will brake the interfaces one more time to use componets, i.e., `Joint` and `Sensor` classes, they should keep using older version for now. I mentioned that in a [issue comment](https://github.com/ros-planning/moveit2/issues/283#issuecomment-695782913)

This issue should remind us to contact the MoveIt guys when we have fixed interfaces (or even better we could submit PR with the solution).

jordan-palacios, 2020-09-22:
Ah, I missed that one. Thanks for handling it.

destogl, 2020-12-12:
I just notified MoveIt guys that API is fixed. (we are at least not expecting to do major changes for Foxy)

bmagyar, 2021-02-16:
Amen. This red flag is not required anymore ;)

closed 2021-02-16 by bmagyar