ros-controls/ros2_control #160: Remove ControllerLoaderInterface

bmagyar, 2020-09-23:
https://github.com/ros-controls/ros2_control/pull/139#discussion_r489310568

mateus-amarante, 2020-10-06:
I would like to get deeper into ros2, so I propose to start solving good first issues. May I get this one?

I only have some questions about it:
* Should `ControllerLoaderPluginlib` exist (as a`ControllerLoader`) or can it be  `pluginlib::ClassLoader<controller_interface::ControllerInterface>` directly?

* Does the list of loaders inside the `ControllerManager` make sense? Can I remove it too?

Karsten1987, 2020-10-06:
#175 is a first start to do that. It needs a bit more love though. If you don't mind, please feel free to take over that PR and either open a PR against that branch or cherry-pick the existing commits onto your own branch and open a new PR from your branch.

Happy to hand this one over :)

mateus-amarante, 2020-10-06:
I didn't see #175. You have already done most of the update :)

I'll finish it.

commit 96f075f3 (2020-10-07): Finish Fix #160 (#185)

* Fix reload_controller_libraries_srv test

* Remove remaining comments

* Remove remaining mock
PR #175 (2020-10-08): remove controller loader abstraction layer
fixes #160
closed 2020-10-08 by bmagyar