ros-controls/ros2_control #164: Implement ResourceManager based on Components

Karsten1987, 2020-09-25:
With the newly introduced components, the requirements for the resource manager are as follows:

* Initially parse the URDF and its components.
* Allocate initial memory for each of these components - these components are fix on runtime.
* Provide Resource Loaning mechanism:
  * "Handles" to components can be uniquely loaned/assigned to controllers.
  * Components are automatically released when controller is shutdown/unloaded.
  * Controller Manager iterates over all components for read/write. 

For more details refer to https://github.com/ros-controls/roadmap/blob/master/design_drafts/components_architecture_and_urdf_examples.md

destogl, 2020-10-26:
Can we rename "Handle" to "Interface".   We are actually talking all the time about Robot-Interfaces (position, velocity, etc...)

The reason I am against the handles because it is confusing to me to access the robot through them. I am missing some clarity.

Is it me, or are there also similar opnions?

Karsten1987, 2020-10-26:
I am in favor of renaming them to interface. That'll go along with the nomenclature we've introduced in the URDF. I'll change #203 to use `CommandInterface` and `StateInterface`. 

What do you think?

PR #236 (2020-12-01): Use resource manager
Last part of #164 and closes #164 

Currently sits on top of #216 and addresses the comment in [here](https://github.com/ros-controls/ros2_control/pull/216#discussion_r518208708).

To start the node one can pass in a URDF as such:
```
ros2 run controller_manager ros2_control_node --ros-args -p update_rate:=10 -p "robot_description:=`cat install/test_robot_hardware/share/test_robot_hardware/test_robot_hardware.urdf`"
```
closed 2020-12-01 by bmagyar