ros-controls/ros2_control #172: Why I can not find libraries after clean build

destogl, 2020-10-01:
I have the following problems:

1. [ERROR] [launch]: Caught exception in launch (see debug for traceback): package 'controller_manager' found at '/home/stogl/workspace/ros2_ws_eloquent/install/controller_manager', but libexec directory '/home/stogl/workspace/ros2_ws_eloquent/install/controller_manager/lib/controller_manager' does not exist
  - `controller_manager` is in '/home/stogl/workspace/ros2_ws_eloquent/install/controller_manager/lib'

2. [ERROR] [launch]: Caught exception in launch (see debug for traceback): executable 'ros2_control_node' not found on the libexec directory '/home/stogl/workspace/ros2_ws_eloquent/install/controller_manager/lib/controller_manager'
  - why is node search in `lib` folder? 
  - node is in `bin` folder

The configuration in  `CMakeLists.txt` seams to be fine... or we need to add `${PROJECT_NAME}` variable to it?

Karsten1987, 2020-10-01:
Every node has to be installed into `../lib/<pkg_name>` in order for the `ros2cli` tools to pick it up. The correct CMake line would be something along the lines of `install(ros2_control_node DESTINATION lib/${PROJECT_NAME}`. See [this example](https://github.com/ros2/examples/blob/master/rclcpp/topics/minimal_publisher/CMakeLists.txt#L30).

destogl, 2020-10-02:
I would to leave this reopened, because we do not have such defintion in almost any CMakeLists.txt

As you can see it also does not work with the controller_manager library.

I will do a PR with this changes.

dignakov, 2020-10-09:
I think that definition only needs to be changed/added to nodes/executables that would be run through `ros2 launch` or `ros2 run`

This should be the right way to do it?
```
# 1st install call for libraries, and executables that should be on PATH
install(TARGETS controller_manager
  RUNTIME DESTINATION bin
  LIBRARY DESTINATION lib
  ARCHIVE DESTINATION lib
)
```
I think the only node so far is `ros2_control_node` , which would need to be in `lib/${PROJECT_NAME}`
```
# 2nd install call for things that ros2 run, and ros2 launch should be able to find
install(TARGETS ros2_control_node 
  RUNTIME DESTINATION lib/${PROJECT_NAME}
)
```

Karsten1987, 2021-02-25:
I am closing this issue (again) as the thing should be resolved now.

closed 2021-02-25 by Karsten1987