ros-controls/ros2_control #174: Realtime safe access to read & write components

Karsten1987, 2020-10-01:
The current component interfaces are not designed for realtime access. I personally see two ways of solving this, both come with some drawbacks:

* expose internal data pointer to the component's internal memory allocation. The idea is equivalent to `std::data` (https://en.cppreference.com/w/cpp/iterator/data) which is implemented by pretty much all STL containers.
 PRO: No need to adapt a complex interface and basically holds for all component types.
 CON: The generic interface doesn't allow for easy memory handling. A `double *` to a complex type(e.g. image or ft-sensor) might be cumbersome to work with.

* don't expose any memory accessors at all on the interface and let each controller request a handle to the specialized component. I could envision a function as such `template<class ComponentT> ComponentT& get_handle()` which in return delivers a precise instance of the loaded component.
 PRO: Memory optimal access to internal data representation. Each component might have optimal functions to treat data.
 CON: A controller is basically required to work with a certain type of component. That is, I can't easily interchange a component with implements a FTSensor with a component for an IMU or such. 

I am tending to the latter as I don't necessarily see the CON as a real deal-breaker. I expect to still have full interoperability between different implementations of the same hardware type. That is, I expect that FTSensorVendorA implements the same sensor component as FTSensorVendorB.

destogl, 2020-10-06:
Here a proposal to make the management of interfaces/handes access simpler from the controller perspective.

1. ControllerManager reads `joint_names` and `interfaces` parameters from the parameter server
  - the parameters are than in the controller_manager namespace
2. ControllerManager ask ResourceManger for confilicts and does not initialize the controller if there are any
3. ControllerManager provide list of `Joint` and `Sensor` components to the `init`-function of the ControllerInterface

The proposal could be tested after #147, #170 and #171 are merged.

Karsten1987, 2020-10-06:
I have an implementation ready for my second proposal which I would love to bring into a PR, but I guess it needs 147 to be merged beforehand. 
alternatively, I could update #165 to reflect these changes.

destogl, 2020-10-07:
@Karsten1987  do you mean the second proposal from your [comment](https://github.com/ros-controls/ros2_control/issues/174#issue-713148056)? This also depends on #171 am I right?

I would go as discussed on Monday, we proceed with #147, #170 and #171 and than close #165 .

closed 2021-02-18 by Karsten1987