ros-controls/ros2_control #191: joint_limits_interface should not wrap JointHandle in shared_ptr

bmagyar, 2020-10-10:
@bmagyar @Karsten1987 @destogl In this refactor for #134 I went with JointHandle arguments wrapped in  std::shared_ptr<>, should these have been without shared_ptr? This was due to a commit I was using as a reference that had shared_ptr, but after reviewing more code such as robot_hardware.cpp I dont see any reason to use shared_ptr here, the JontHandle by value would work fine and probably more compatible. Thoughts?

_Originally posted by @guru-florida in https://github.com/ros-controls/ros2_control/pull/181#issuecomment-706319954_

bmagyar, 2020-10-10:
@guru-florida please go ahead and make these changes on a new PR, I've merged the old one.
You may want to branch off from a fresh checkout of master, we squash-merge PRs.

guru-florida, 2020-10-10:
Will do. I can take care of #188 and #189 at the same time. Would you like them as individual PRs or just one?

bmagyar, 2020-10-10:
I've already submitted for #189 and #190  in #193 and #192. 

Happy for you to do #188, just leave a note that you are doing it so nobody else starts it. It's fine to mash them up in the same PR :+1:

guru-florida, 2020-10-13:
@bmagyar PR #194 fixes issue #191 and #188. Please, review caveat in PR comment about hardware_interface.

destogl, 2020-10-21:
@guru-florida this seams to be solved in #194. Is it OK to close the issue?

guru-florida, 2020-10-21:
Yes, this and #188 can be closed.

closed 2023-05-09 by bmagyar