ros-controls/ros2_control #208: consolidate test material

Karsten1987, 2020-10-23:
many tests are requiring some sort of mocked robot material, whether it's the actual hardware or a URDF.
Right now, these things are copied across multiple repositories and should be ideally consolidated into a single location, most likely their own package in the style of `ros2_control_test_material`.

destogl, 2020-10-24:
Thanks for opening this!

bmagyar, 2020-10-24:
I'd propose to move `test_robot_hardware` in there too or whatever is the equivalent in the future. How about `ros2_control_test_assets`?

destogl, 2021-01-06:
I started to work on this in #247, but there were some issues with dependencies (see more in the discussion there).

Therefore, I will proceed as follows:
- [ ] move test hardware components from `test_robot_hardware` to `hardware_interface` package (#288)
- [ ] add `description.hpp` file to the new `ros2_control_test_assets` package. This file is already used by all controllers. I will check how to use it for tests in the rest of the framework.

bmagyar, 2021-02-16:
I'd consider this done, please reopen if you have other points to raise @destogl

closed 2021-02-16 by bmagyar