ros-controls/ros2_control #210: change min/max values to double

Karsten1987, 2020-10-23:
The current component_info declares their min and max values as strings. I am not really sure why that's the case, but I think this should be changed to doubles.

bmagyar, 2020-10-24:
It is considered an optional argument in the implementation, I guess it was an easy way to communicate that. 
A better solution would be `std::optional<double>` or `boost::optional<double>` but I take it none of that is really an option for us ATM. Is there an `rclcpp::optional` that's not using boost and doesn't need C++17? Or can we just move to C++17 now?

destogl, 2020-10-24:
The idea was to provide different interface types, so that the concrete component would parse the data into a specific type. But since we are being more specific now, it can make sense to support only doubles.

Will we ever have an interface which uses any other data than double? Some tools or something maybe? Do you have any idea?

Let's just move to C++17...

Karsten1987, 2020-10-25:
I guess that's reason enough for me to justify the use of a string here. Thanks for the clarification.

closed 2020-10-25 by Karsten1987