ros-controls/ros2_control #213: controller_interface namespace

ahcorde, 2020-10-26:
I'm facing an issue in `gazebo_ros2_control` https://github.com/ros-simulation/gazebo_ros2_control/issues/37. The `controller_interface` is not setting up the node namespace.

See this line of code: https://github.com/ros-controls/ros2_control/blob/master/controller_interface/src/controller_interface.cpp#L29

destogl, 2020-10-29:
Is #216 solving your issue? Check this line of code: https://github.com/ros-controls/ros2_control/pull/216/files#diff-ef3035625d6e5c566b8d29dc255f3297c6a53186b8ebf048e639198e0df937deR30

This code worked in my tests with "components" concept. It also solves the issue with undeclared parameters #168

ahcorde, 2020-10-29:
Not really, In gazebo_ros2_control, I need to create an instance of this class.  I can't see in these lines how the namespace is included.

Maybe we can create another init() method with the namespace as an argument ?

https://github.com/ros-controls/ros2_control/pull/216/files#diff-ef3035625d6e5c566b8d29dc255f3297c6a53186b8ebf048e639198e0df937deR29-R32

destogl, 2020-10-31:
Oh sorry, I misinterpreted your question...

Yes we could do this

bmagyar, 2021-02-16:
@ahcorde is this still an issue?

Briancbn, 2021-02-16:
@bmagyar Thanks for following up on this.

This is still an issue. Although now with node name included inside the `forware_command_controller` subscription topic as `~/command` solves some of the problem, `/joint_states` published by `joint_state_controller` still cannot be remapped with <namespace> argument within the gazebo plugin.

I think the problem is that `ControllerManager` constructor doesn't call the node constructor that will pass along the namespace. I believe another overload constructor calling the explicit constructor of `Node` below should solve the problem.

Node constructor that pass along the namespace:
https://github.com/ros2/rclcpp/blob/18b112fcce90ef41fb2a312dda59508f2164dc5e/rclcpp/include/rclcpp/node.hpp#L97-L101

@ahcorde please help me elaborate if you have much better insight on this.

bmagyar, 2021-02-16:
The global name for `/joint_states` is pretty much a standard location. Why do you need to remap it?

Also, I wonder if this is this something that can be solved entirely on the gazebo side now?

Briancbn, 2021-02-16:
> The global name for `/joint_states` is pretty much a standard location. Why do you need to remap it?

Two manipulator with separate moveit setup in gazebo to represent two independent workcell, which means controller manager and controllers under different namespace. Any suggestion?

> Also, I wonder if this is this something that can be solved entirely on the gazebo side now?

As new nodes and separated executor are created within the ros2_control,  and separated executor are created for controller_manager as well, Gazebo plugins at the moment can't seem to do any remap or set any namespace. I understand that `init(const & controller_name, const & controller_namespace = "")` will break APIs and also downstream inheritance override in `ros2_controllers`, but currently it seems that is the only solution.

destogl, 2021-02-21:
> > The global name for `/joint_states` is pretty much a standard location. Why do you need to remap it?
> 
> Two manipulator with separate moveit setup in gazebo to represent two independent workcell, which means controller managers and managed controllers under different namespace. Any suggestion?

This shouldn't actually matter at all. Are you using prefixes for the joint names for your two manipulators? IMO this is the proper way. You actually want to have a global topic named '/joint_names' receiving data about all joint in the system. Even from different robots and other devices.

gavanderhoorn, 2021-02-22:
> You actually want to have a global topic named '/joint_names' receiving data about all joint in the system. Even from different robots and other devices.

That doesn't mean that not having a global `/joint_states` topic should not be supported.

It doesn't seem like "nice ROS-nodeship" to publish to global, fully resolved names, as common as they are in their usage.

> Are you using prefixes for the joint names for your two manipulators? IMO this is the proper way. 

Having completely separated graphs within the same application is not such an uncommon occurrence.

bmagyar, 2021-02-22:
Thanks for this feedback guys.

So let's see a list of actions to solve this:
* joint_state_controller parameter to use or ignore the local namespace it is in: I feel that we should still maintain the use of the global `/joint_states` topic by default but a simple flag setting would help anyone wanting to do a multi-robot system constrain it to the namespace.
* @Briancbn what can we do about the other issues in gazebo?

gavanderhoorn, 2021-02-22:
>     * joint_state_controller parameter to use or ignore the local namespace it is in: I feel that we should still maintain the use of the global `/joint_states` topic by default but a simple flag setting would help anyone wanting to do a multi-robot system constrain it to the namespace.

is the issue really specific to multi-robot systems?

Publishing to a fully resolved name seems like it should not be the default.

oKermorgant, 2021-10-14:
Hi, 

Any news on this issue?

I am trying to convince my team to use ros2_control for a fleet of autonomous underwater vehicles. In this case each robot subscribes to its own namespaces joint_states (camera tilt, thrusters, fins, etc.) and for sure we do not want that information to be broadcast to all other vehicles.

It could be the same for several identical manipulators in a factory: joint names would the same for all robots, but _joint_states_ appear in different namespaces that all have their own _robot_state_publisher_.

Briancbn, 2021-10-31:
> I am trying to convince my team to use ros2_control for a fleet of autonomous underwater vehicles. 

Hi @oKermorgant, are you using hardware or Gazebo simulation? The above issue only affects gazebo simulation as `controller_manager` has to launched through a Gazebo plugin, not a ros2 launch file.

For hardware, simply configuring your launch file to include the `namespace='namespace'` for the `controller_manager` node would work. Don't forget to set the controller_manager node name using the `--controller-manager` option as well if you want to use the spawner. We have tried with our hardware and it works.

> In this case each robot subscribes to its own namespaces joint_states (camera tilt, thrusters, fins, etc.) and for sure we do not want that information to be broadcast to all other vehicles.

If you are using a launch file for the `controller_manager`, a simple remap with global namespace (with a '/' in front) should solve your issue. However, it could not be done in Gazebo when `controller_manager` is loaded as a plugin.

oKermorgant, 2021-11-02:
Hi @Briancbn, indeed we use it with Gazebo and on the real robots.
Anyway for now we have implemented (yet another) home-made PID inspired from ros1's control_toolbox.

dreenott, 2023-10-09:
This is still an issue from my testing!