ros-controls/ros2_control #215: JointStateHandle not available

kong2983, 2020-10-28:
I am configuring HardwareInterface using JointStateHandle. As in the example, 
I tried to call hardware_interface::JointStateHandle using the latest package, but I got an error saying that the package could not be found. 
I noticed that the JointState Handle disappeared from the latest package and the JointHandle package was created.
Should I change the existing JointStateHandle to JointHandle?
And, where can I find an example of this change?

Thank you for your reply.

destogl, 2020-10-29:
Hi @kong2983,

You can do this, but please keep in mind that the interface will change again soon. So if you are doing only some testing, you can use older version of start incremental switch. It's up to you.

guru-florida, 2020-10-30:
Hi @kong2983, 

The process to switch JointStateHandle code to use new JointHandle is basically:

1. Loop through your joints calling RobotHardware::register_joint(JointName, JointInterfaceName). JointInterfaceName is going to be one of "position", "velocity" or "effort" for states or "position_command", "velocity_command", etc for command handles.
2. Loop through your joints again and retrieve handles to the registered joints using RobotHardware::get_joint_handle(handle). The handle arg is a JointHandle object with JointName and JointInterfaceName already set. So the get_joint_handle looks up the registered joint+interface and then binds the handle to the value reference.
3. Rename all the handle.getPosition() or getVelocity() etc to just get_value()

FYI Dont try to combine step 1 and step 2 in the same loop or you'll encounter the situation in #212.

If you are going to store your handles in a collection like vector you may want to resize your vector from the start using this since JointHandle doesnt have a default constructor.
`joint_pos_stateh_.resize(n_dof_, hardware_interface::JointHandle("position"));

You can see an example of this in the [Gazebo ros2_controls PR34](https://github.com/ros-simulation/gazebo_ros2_control/pull/34
) which is pending here and specifically see the [default_robot_hw_sim.cpp file](https://github.com/ros-simulation/gazebo_ros2_control/blob/ahcorde/update/ros2_control/gazebo_ros2_control/src/default_robot_hw_sim.cpp).

closed 2020-12-02 by bmagyar