ros-controls/ros2_control #229: controller_manager libexec directory not found

AndyZe, 2020-11-09:
When launching the `controller_manager` from a launch.py file like this:

```
    return LaunchDescription([
      Node(
        package='controller_manager',
        executable='ros2_control_node',
        parameters=[robot_description, robot_controller_file],
        output={
          'stdout': 'screen',
          'stderr': 'screen',
          },
        )
    ])
```

I get the error:

`exception=SubstitutionFailure("package 'controller_manager' found at '/home/andy/ws_ur2/install/controller_manager', but libexec directory '/home/andy/ws_ur2/install/controller_manager/lib/controller_manager' does not exist")`

Screenshot of my install directory attached.

[image]

AndyZe, 2020-11-09:
I suppose this is because the executable is not ready yet, and I just need to wait?

destogl, 2020-11-18:
This is probably related to #172. 

Can you check it?

bmagyar, 2021-02-16:
Is this still an issue? Any different from #172?

closed 2021-02-25 by Karsten1987