ros-controls/ros2_control #235: Avoid heap allocation for pluginlib in controller manager

Karsten1987, 2020-11-11:
I tried to avoid using a shared pointer (or generally prevent heap allocation) for the pluginlib and I ran into a segfault when trying to reload the controller libraries.

error:
```
[ RUN      ] TestControllerManagerSrvs.reload_controller_libraries_srv
[INFO] [1605123552.745224493] [test_controller_manager]: Controller manager: reloaded controller libraries for 'controller_interface'
[INFO] [1605123552.745338606] [test_controller_manager]: Loading controller 'test_controller_name'

Process 70670 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x38)
    frame #0: 0x0000000100bd9f47 libclass_loader.dylib`class_loader::MultiLibraryClassLoader::getAllAvailableClassLoaders() const + 247
libclass_loader.dylib`class_loader::MultiLibraryClassLoader::getAllAvailableClassLoaders:
->  0x100bd9f47 <+247>: movq   0x38(%r13), %rax
    0x100bd9f4b <+251>: movq   %rax, (%r12,%rbx,8)
    0x100bd9f4f <+255>: leaq   0x8(%r12,%rbx,8), %rbx
    0x100bd9f54 <+260>: testq  %r15, %r15
(lldb) r
There is a running process, kill it and restart?: [Y/n] y
Process 70670 exited with status = 9 (0x00000009)
Process 71014 launched: '/Users/karsten/workspace/ros2/ros2_control_ws/build/controller_manager/test_controller_manager_srvs' (x86_64)
Running main() from gmock_main.cc
```

All other tests passed correctly, but it seems that replacing
```c++
  // Force a reload on all the PluginLoaders (internally, this recreates the plugin loaders)
  loader_ = std::make_shared<pluginlib::ClassLoader<controller_interface::ControllerInterface>>(
    kControllerInterfaceName, kControllerInterface);
```
with 
```c++
  // Force a reload on all the PluginLoaders (internally, this recreates the plugin loaders)
  loader_ = pluginlib::ClassLoader<controller_interface::ControllerInterface>(
    kControllerInterfaceName, kControllerInterface);
```

yields to some internal error. Don't really know why.

@v-lopez any idea?

Karsten1987, 2020-11-11:
At the same time, I am seeing the following:
```
[ RUN      ] TestControllerManagerSrvs.reload_controller_libraries_srv
[INFO] [1605126015.192235243] [test_controller_manager]: Controller manager: reloaded controller libraries for 'controller_interface'
[INFO] [1605126015.192349011] [test_controller_manager]: Loading controller 'test_controller_name'

[INFO] [1605126015.208069530] [test_controller_manager]: Controller manager: Stopping all running controllers
[INFO] [1605126015.208127635] [test_controller_manager]: Empty start and stop list, not requesting switch
Warning: class_loader.ClassLoader: SEVERE WARNING!!! Attempting to unload library while objects created by this loader exist in the heap! You should delete your objects before attempting to unload the library or destroying the ClassLoader. The library will NOT be unloaded.
         at line 132 in /Users/karsten/workspace/ros2/ros2_foxy/src/ros/class_loader/src/class_loader.cpp
[INFO] [1605126015.261173786] [test_controller_manager]: Controller manager: reloaded controller libraries for 'controller_interface'
[INFO] [1605126015.262782316] [test_controller_manager]: Loading controller 'test_controller_name'

[ERROR] [1605126015.275837579] [test_controller_manager]: Controller manager: Cannot reload controller libraries because there are still 1 controllers running
[INFO] [1605126015.275995108] [test_controller_manager]: Controller manager: Stopping all running controllers
Warning: class_loader.ClassLoader: SEVERE WARNING!!! Attempting to unload library while objects created by this loader exist in the heap! You should delete your objects before attempting to unload the library or destroying the ClassLoader. The library will NOT be unloaded.
         at line 132 in /Users/karsten/workspace/ros2/ros2_foxy/src/ros/class_loader/src/class_loader.cpp
[INFO] [1605126015.381138705] [test_controller_manager]: Controller manager: reloaded controller libraries for 'controller_interface'
[       OK ] TestControllerManagerSrvs.reload_controller_libraries_srv (270 ms)
```

which makes me wonder why this test is marked as `passed` if clearly something goes wrong there.

v-lopez, 2020-11-13:
Regarding the warning in the test, it is due to the test keeping an instance of the controller [here](https://github.com/ros-controls/ros2_control/blob/master/controller_manager/test/test_controller_manager_srvs.cpp#L219), that is used [after the reload](https://github.com/ros-controls/ros2_control/blob/master/controller_manager/test/test_controller_manager_srvs.cpp#L249) to make sure that the `force_kill` indeed cleans up the controller before destroying the instance and decreasing the shared_ptr `use_count` by 1.

If I reset that shared_ptr before calling the reload, and comment those asserts, the warning disappears.

I could replace the shared_ptr with a weak_ptr and keep one of the asserts (the one about the use_count).
For the unconfigured state assert, I would have to spend a bit more time to make the test controller store the state changes somewhere so I could check after the instance is deleted.

Regarding removing the heap allocation, pluginlib is heap memory all the way down, I can give a look into why changing that shared_ptr to regular storage crashes, but heap allocation is going to happen inside anyway on the constructor.

bmagyar, 2020-12-02:
I agree it'd be very hard to remove heap allocation for plugins... What's the verdict on this?

v-lopez, 2020-12-02:
I'd vote for closing this because of what I stated at the end of my previous message. 
Also I'd focus on what happens in the update() function, which is the RT critical code.

closed 2020-12-02 by bmagyar