ros-controls/ros2_control #244: Extend list_interfaces service with interface claimed status

destogl, 2020-11-19:
I know it's a PITA but I'd strongly suggest renaming this service. We already have the name "controller interface" for a class, also it doesn't actually capture what's happening here, there's not even a controller at play in this context.

I suggest something along the lines of `ListJointInterfaces` but I wonder if this is strongly required at this stage... I think it may be useful to also denote whether these interfaces are claimed at the moment, perhaps by which controller, etc. I think that from the resource manager PR perspective, this addition is a bit of a feature creep that can lead to a lot more creeping as you can tell by my comment.

Final thought: I'd actually suggest moving this feature to a new PR and we can discuss details of it there.

_Originally posted by @bmagyar in https://github.com/ros-controls/ros2_control/pull/236#discussion_r526689893_

destogl, 2020-11-19:
Also mentioned in [#236 (comment)](https://github.com/ros-controls/ros2_control/pull/236#discussion_r526919795)

bmagyar, 2021-03-24:
Done

closed 2021-03-24 by bmagyar