ros-controls/ros2_control #249: Replace ControllerInterface LifecycleNode with Node

v-lopez, 2020-11-25:
Controllers are not regular nodes, they cannot be activated or deactivated from anyone but the controller manager, for many reasons:

1. If we use the `ResourceManager`, we need the `ControllerManager` to assign hardware interfaces. Assignment occurs [before activation](https://github.com/ros-controls/ros2_control/pull/236/files#diff-51a3ffe225a0370e8b9026ea1cb8f0cf34fd5bf6547a239ce8418f445e22a094R654).
2. Deactivating, sometimes requires the controller to send a last command to the hardware. For instance a 0 velocity command or a hold position command, for a safe shutdown. This should not be done from a random thread after deactivate was called from someone else in the system, it can only be done in the update() function of the `ControllerManager`, where it is thread safe to write this last command to the `CommandInterfaces`
3. In general, any call to configure/activate/deactivate/cleanup that can be triggered externally, will end up in a random thread causing concurrency issues when the controller is being used by the RT thread.

As a summary, I believe it is dangerous to leave the state transitions open for anyone to trigger when they want, since they require coordination with the `ControllerManager`.
I don't know a way of disabling that, so my proposal is to replace `LifecycleNode` with `Node`. 
And if we do that, we might as well remove the inheritance from `LifecycleNodeInterface` in `ControllerInterface` and just declare virtual `configure`, `activate`, `deactivate` and `cleanup` methods without the `previous_state` arg, since our state machine is clear that only can be: configure -> activate -> deactivate -> cleanup, or configure -> cleanup.

v-lopez, 2020-11-25:
Another option is to embrace the whole lifecycle thing, but then I believe we need at least to:

1. Ensure that the lifecycle callbacks are executed on the RT thread (whether they are RT-safe or not is another issue) or guarantee that their execution doesn't overlap with the update method.
2. Allow the controllers to claim hardware interfaces themselves, without the controller_manager being involved, so on a Controller's `activate` it can attempt to acquire all the interfaces it needs, and if it fails it can return that the transition failed.

destogl, 2020-11-25:
> And if we do that, we might as well remove the inheritance from `LifecycleNodeInterface` in `ControllerInterface` and just declare virtual `configure`, `activate`, `deactivate` and `cleanup` methods [...]

If we do that I vote for `ros_control` names: `activate` == `start`; `deactivate` == `stop`

> Allow the controllers to claim hardware interfaces themselves, without the controller_manager being involved, so on a Controller's activate it can attempt to acquire all the interfaces it needs, and if it fails it can return that the transition failed.

I don't think we should do this.

v-lopez, 2020-11-25:
I agree with you on the name thing, it would reduce confusion with the lifecycle states.

bmagyar, 2020-11-25:
Thanks for looking deeper into this Victor. 
I agree that the external controller lifecycle node interfaces are problematic, especially now that with #236 we are eliminating the connection from controller to RobotHardware/ResourceManager. 

The LifecycleNode *reporting* interface and API still have value. My slightly lazy question is: can we have LifecycleNodes *without* the external service interfaces? If we can reduce this to only report states and transitions and be only accessible through the C++ API, we wouldn't need to change much. I am lacking insight to answer this though.

In case we cannot hide the service interfaces, in the interest of delivering a release, I vote for putting investigating how to make these interfaces safe on the TODO list and revert back to regular Nodes and the `ros_control` nomenclature @destogl mentioned.

Karsten1987, 2020-11-25:
> I don't know a way of disabling that, so my proposal is to replace LifecycleNode with Node.

If so, why bother with Nodes in the first place? If you change from lifecycle node to node, you still have external influence over it, such as their subscriptions, parameter callbacks and whatnot. I am not sure we win much with this.

> Ensure that the lifecycle callbacks are executed on the RT thread (whether they are RT-safe or not is another issue) or guarantee that their execution doesn't overlap with the update method.

That's where I see the main focus for now. I think we have to expose the callback groups more explicit and make sure that every controller can add services to either an rt-safe or rt-unsafe callback thread. But I also feel like this needs more thinking and ideally some sort of design doc.

> Allow the controllers to claim hardware interfaces themselves, without the controller_manager being involved, so on a Controller's activate it can attempt to acquire all the interfaces it needs, and if it fails it can return that the transition failed.

That's still the case, but it's slightly inverted. The controllers are declaring all the interfaces it wants to claim and the controller manager assigns them. That's to say, the transaction is not even going to be triggered if the resources don't match. I like that design quite a bit as it shifts the focus from the controller to the controller manager.

> can we have LifecycleNodes without the external service interfaces? 

We can do this on an rcl level, but it's not exposed on C++ level.
https://github.com/ros2/rclcpp/blob/master/rclcpp_lifecycle/src/lifecycle_node_interface_impl.hpp#L99
I am however not so sure if that actually makes sense. If we get rid of the lifecycle services, then one could also question why we have lifecycle nodes and further ask why controllers are their own nodes in the first place.

v-lopez, 2020-11-25:
>  If so, why bother with Nodes in the first place? If you change from lifecycle node to node, you still have external influence over it, such as their subscriptions, parameter callbacks and whatnot. I am not sure we win much with this.

The idea of keeping Nodes was so they could still define params in their own namespaces, as well as "private" publisher, subscribers, etc... 
The problem in my opinion is the Lifecycle part of it, not the Node.

> That's still the case, but it's slightly inverted. The controllers are declaring all the interfaces it wants to claim and the controller manager assigns them. That's to say, the transaction is not [...]

I like this as well, but then it renders the activate lifecycle API useless from the outside, since you must go through start_controllers, or you won't be assigned those resources.

destogl, 2020-11-25:
> > If so, why bother with Nodes in the first place? If you change from lifecycle node to node, you still have external influence over it, such as their subscriptions, parameter callbacks, and whatnot. I am not sure we win much with this.
> 
> The idea of keeping Nodes was so they could still define params in their own namespaces, as well as "private" publishers, subscribers, etc...
> The problem in my opinion is the Lifecycle part of it, not the Node.

I agree with @v-lopez on this. There is still value for the controller to be their nodes, exactly to have their services and parameters.

This obviously needs quite more discussion and thinking. Should we for sake of time "fall-back" to regular nodes and then discuss other options for galactic?

bmagyar, 2020-11-26:
With this in mind I'd opt for falling back to Nodes for Foxy, let's see how those work with RT issues and re-evaluate our options for Galactic.

v-lopez, 2020-11-26:
So we merge #236, the ones in ros2_controllers for using resource_manager, and then we replace lifecyclenode with node?

destogl, 2020-11-27:
> So we merge #236, the ones in ros2_controllers for using resource_manager, and then we replace lifecyclenode with node?

Yes, that's the plan. (We will also merge #240 after #236)
@bmagyar would like to have controller working before merge.

v-lopez, 2020-12-02:
Regarding this, would you like me to keep as much as possible the API of the original implementation.

We could even keep LifecycleNodeInterface inheritance, but we would need to define some `configure`, `activate`.. methods like the ones that would be retrieved from LifecycleNode, and connect these methods to the `on_configure` overrides from the controllers.

destogl, 2020-12-03:
Would it make sense to create a 'LifecycleInterface' from which 'ControllerInterface' inherits.

So we create a compatibility layer similar to the Lifecycle node implementation.

v-lopez, 2020-12-03:
It already inherits from LifecycleInterfaceNode which has the virtual methods for `on_configure`, `on_activate`, etc...
These callbacks are connected to a member LifecycleNode's `configure`, `activate`,... 
I don't think the API for the controllers will change in this regard, they still have to put there their code in these callbacks.

What will change is that now there will be no `configure` method provided by the LifecycleNode, so we need to provide an alternative to `get_lifecycle_node()->configure`.

bmagyar, 2021-02-16:
Closing as we've resolved this for Foxy.

closed 2021-02-16 by bmagyar