ros-controls/ros2_control #250: Use constants instead of strings for standard interfaces

destogl, 2020-11-26:
> looks good to me, even though I'd like to see these values then applied to other packages as well - for consistency.

I will do that, as soon as we clarify exact names.

_Originally posted by @destogl in https://github.com/ros-controls/ros2_control/pull/241#discussion_r529742582_

destogl, 2021-03-24:
Check if there are any of the following hard-coded string:

- "position"
- "velocity"
- "acceleration"
- "effort"

Those are defined as constants [here](https://github.com/ros-controls/ros2_control/blob/master/hardware_interface/include/hardware_interface/types/hardware_interface_type_values.hpp) and should be used instead.

mateus-amarante, 2021-04-07:
I can do that if there is no one working on it yet

bmagyar, 2021-04-07:
Go for it!

PR #376 (2021-04-11): Replace standard interfaces' hard-coded strings by constants
Replace the standard interfaces hard-coded strings "position", "velocity", "effort", and "acceleration" by the constants defined in [hardware_interface/types/hardware_interface_type_values.hpp](https://github.com/ros-controls/ros2_control/blob/master/hardware_interface/include/hardware_interface/types/hardware_interface_type_values.hpp).

Fix #250
closed 2021-04-11 by bmagyar