ros-controls/ros2_control #251: Remove controller_name from init function in ControllerInterface

destogl, 2020-11-26:
Reviewing the current controller portings in ros-control/ros2_controllers repository I realized, that this name argument is obsolete. Since controllers are `LifecycleNode`s they have access to their own name through the node name.

```suggestion
  init();
```

_Originally posted by @destogl in https://github.com/ros-controls/ros2_control/pull/236#discussion_r530619435_

Should be done with changes in #246

bmagyar, 2021-08-31:
@destogl where do we stand on this nowadays?

bmagyar, 2021-09-15:
@destogl @bailaC ping

destogl, 2021-09-20:
> @destogl where do we stand on this nowadays?

I think this obsolete having `on_init` now in the controllers. Also as discussed with @bailaC this is not really implemntable and it doesn't make much sense anymore. Therefore I am closing this.

closed 2021-09-20 by destogl