ros-controls/ros2_control #252: Create publisher/subscriber when inheriting SystemInterface

livanov93, 2020-11-30:
While watching the [hardware interface example](https://github.com/ros-controls/ros2_control_demos/blob/add_rrbot_system_position_joints/ros2_control_demo_hardware/include/ros2_control_demo_hardware/rrbot_system_position_only.hpp), I wondered how can publishers and subscribers be created when inheriting from [SystemInterface](https://github.com/ros-controls/ros2_control/blob/use_resource_manager/hardware_interface/include/hardware_interface/components/system_interface.hpp). 
As I could understand there is no getter to fetch the rclcpp::Node from the [ros2_control_node](https://github.com/ros-controls/ros2_control/blob/use_resource_manager/controller_manager/src/ros2_control_node.cpp). Is there?

My only idea is to have an [executor](https://github.com/ros2/rclcpp/blob/master/rclcpp/include/rclcpp/executors/single_threaded_executor.hpp) and a [node](https://github.com/ros2/rclcpp/blob/master/rclcpp/include/rclcpp/node.hpp) as members of the class that inherits SystemInterface, register publisher/subscirber and to create another thread within the class that will spin the thing. 

What am I missing here?

destogl, 2020-11-30:
May I ask why do you need publisher ans subscribers in there? I would use those always through controller, so you know that your real-time constraints are respected as much as possible.
If you need services, that's another beast.

livanov93, 2020-12-01:
I would like to share a flag that is fetched from the robot driver within system hardware interface. It refers to the current state of robot's control, if it accepts commands from ROS side or not.

bmagyar, 2020-12-02:
@destogl I can also imagine a non-realtime-safe setup may want to have their data leave ros2_control through topics, I've done similar things with ROS1 and rosserial. 

@livanov93 I think your approach of a new thread is the best way for now, dealing with other parameters through the ros2_control_node executor could make things on that end way too messy, we are trying to keep things very tightly controlled there.

livanov93, 2020-12-02:
@bmagyar What would be the proper way of handling this now, with respecting real-time contraints?

I could return return_type::ERROR from the [write()](https://github.com/ros-controls/ros2_control_demos/blob/f44d20bee525eeb1363c3f27dd4891875a41b27b/ros2_control_demo_hardware/src/rrbot_system_position_only.cpp#L191) method when the robot stops receiving commands from ROS side.

AndyZe, 2020-12-02:
Basically we're just copying the way things were done in ROS1. We may need to rethink this, but we wanted to make service calls to do things like clear robot warnings. I guess the hardware interface may not be the best place to put this stuff.

bmagyar, 2020-12-04:
An ideal solution would be self-contained. If you can clear warnings in start/stop methods as opposed to read/write of your System, that should be fine. You'd still need to spin your own Node probably.

closed 2020-12-09 by livanov93

destogl, 2020-12-10:
I propose to use specialized controllers for now. In the future we have to consider making HW-Interfaces to Nodes. Or something that has it's own parameters and can get callbacks on their change...